solectrus/solectrus #105: Public vs. private mode

ledermann, 2021-02-04:
Public mode:

- No login
- Reduced data presentation - Inverter power only (no consumption)

Private mode:

- Admin login required
- All data, including consumption

Depends on #28

robi-van-kinobi, 2023-01-16:
Hi @ledermann ,

I achieved this with Authelia & Traefik.

This way I was able to delegate Authentication & Authorization to Authelia. Traefik has a middleware which checks if Auth data are correct.

For me this works like a charm. I don't need further User management.

ledermann, 2023-01-17:
@robi-van-kinobi I didn't know Authelia before, thanks for bringing it to my attention, will try this out.

But beside of this: Perhaps it would be useful to open SOLECTRUS **partially** to the public. E.g. show `inverter_power` but not `house_power` and not `wallbox_power`. And maybe don't show prices and costs. This could be useful for stricter privacy. But this is just a rough idea, I don't have plans to implement this in the near future.