solectrus/solectrus #1132: [Feature Request] Connect Solectrus to existing influxdb

tobintax, 2022-06-28:
Hi, 

I already collect data from other pv inverters (e.g. fronius) and write them continously to my influx db. I just need a cool frontend for displaying the data. So just displaying is needed, no database, no collection etc.)

It would be nice to have the possibility to connect solectrus to an existing database. It would make solectrus very useful for other inverters. I think we need a config file to configure the field mapping needed for solectrus to display. E.G. Solectrus needs a value "pv_consumption" and reads that from DB Schema xyz on measurement "pv_fronius_123_consumption". 

Thanks

Tobintax

ledermann, 2023-03-30:
Grundsätzlich benötigt SOLECTRUS eine bestimmte Struktur der Messwerte in einem InfluxDB-Bucket. Wie die Daten da hineinkommen, ist egal. Der Standard-Weg dürfte der SENEC-Collector (für Live-Daten) und der SENEC-Importer (für historische Daten) sein. Entscheidend ist nur, dass `BUCKET`, `MEASUREMENT` und alle `FIELD`s so bezeichnet sind, wie SOLECTRUS es erfordert - welche das sind, lassen sich in der Admin-Oberfläche von InfluxDB ersehen.

Um eine andere, nicht von SOLECTRUS angelegte Influx-Datenbank abzufragen, wäre es denkbar, wie von @tobintax vorgeschlagen, ein Mapping als optionale Konfiguration von SOLECTRUS zu ergänzen. Probleme gibt es aber spätestens dann, wenn es sich nicht nur um andere Bezeichnungen handelt, sondern es auch inhaltliche Unterschiede gibt. Beispiel: SOLECTRUS erwartet die Felder `grid_power_plus` und `grid_power_minus` (beide positiv). Das lässt sich nicht auf ein vorhandenes Feld `grid_power` mappen, sondern da müsste eine Logik hinzukommen, die Berechnungen durchführen kann. Insofern denke ich **nicht**, dass ein simples Mapping ausreichen wird.

Besser finde ich die von @noffycws vorgeschlagene Lösung einer Migration. Das sollte mit Bordmitteln von InfluxDB machbar sein.

1. Admin-Login in die bestehende InfluxDB
2. Neuen Bucket anlegen mit dem Namen "SENEC" (für SOLECTRUS )
3. Im Data Explorer per Query die Daten rüberschieben und dabei umstrukturieren wie nötig. Beispiel:

```flux
from(bucket: "AlterEimer")
  |> range(start: v.timeRangeStart, stop: v.timeRangeStop)
  |> filter(fn: (r) => r["_measurement"] == 'AltesMeasurement")
  |> filter(fn: (r) => r["_field"] == "ErzeugterStrom")
  |> map(fn: (r) => ({ r with _value: r._value, _field: "inverter_power" }))
  |> to(bucket: "SENEC")
```

So ungefähr. In den Queries können Mappings und Berechnungen vorgenommen werden. Man muss sich nur etwas mit der [Flux language](https://docs.influxdata.com/flux/v0.x/) auseinandersetzen :-)

ledermann, 2023-05-29:
Für die Anbindung an Nicht-SENEC-Systeme steht mittlerweile der (noch experimentelle) [MQTT-Collector](https://github.com/solectrus/mqtt-collector) bereit. 

Ich denke, dieser Weg ist sinnvoller. Den oben gemachten Vorschlag, SOLECTRUS an eine bestehende Influx-Datenbank anzubinden, verfolge ich nach jetzigem Stand nicht weiter und schließe daher dieses Issue.

closed 2023-05-29 by ledermann