solectrus/solectrus #1471: Problem in stats year / all years

iot-sle, 2022-11-15:
Error in displaying stat of the first year and all years. 
Strompreis : 01.05.22 0,2268€
                    01.05.21 0,2586€
Einspeisevergütung: 30.11.2021 0,0693 €  

Errorlog:

`I, [2022-11-15T17:10:37.017877 #11]  INFO -- : method= path= format= params={} controller=ApplicationCable::Connection action=disconnect status=200 duration=0.53
I, [2022-11-15T17:10:37.057169 #11]  INFO -- : [172.19.0.1] method=GET path=/stats/year/inverter_power format=html controller=StatsController action=index status=500 duration=16.98 view=0.00 db=0.30 user_agent=Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/107.0.0.0 Safari/537.36
F, [2022-11-15T17:10:37.058771 #11] FATAL -- : [172.19.0.1]   
[172.19.0.1] ActionView::Template::Error (nil can't be coerced into Float):
[172.19.0.1]     10:         - if period == 'now'
[172.19.0.1]     11:           = render DashboardNow::Component.new calculator:
[172.19.0.1]     12:         - else
[172.19.0.1]     13:           = render DashboardRange::Component.new calculator:,
[172.19.0.1]     14:                                                 period: permitted_params[:period],
[172.19.0.1]     15:                                                 timestamp: permitted_params[:timestamp]
[172.19.0.1]     16: 
[172.19.0.1]   
[172.19.0.1] app/services/calculator/range.rb:38:in `*'
[172.19.0.1] app/services/calculator/range.rb:38:in `block in paid'
[172.19.0.1] app/services/calculator/range.rb:37:in `paid'
[172.19.0.1] app/services/calculator/range.rb:55:in `solar_price'
[172.19.0.1] app/services/calculator/range.rb:72:in `profit'
[72.19.0.1] app/components/dashboard_range/component.html.slim:28:in `block in call'
[172.19.0.1] app/components/radial_badge/component.html.slim:10:in `call'
[172.19.0.1] app/components/dashboard_range/component.html.slim:27:in `call'
[172.19.0.1] app/views/stats/index.turbo_stream.slim:13
[172.19.0.1] app/controllers/stats_controller.rb:5:in `index'`

All other dashboards are working. So only the starting year 2021 is crashing. Current year 2022 is working fine. By the way starting date of the photovoltaik was 30.11.2021.
I guess the error was first shown, when I started using the renaul-collector for the first time.

ledermann, 2022-11-15:
Ok, this seems to be caused by a price starting **before** installation date. I will fix this with the next update.

Try this as workaround: 
Edit the price of **01.05.2021**(0,2586€). Change date to **30.11.2021** (installation date). Save.

iot-sle, 2022-11-15:
Thanks for your fast reaction. I confirm the problem is gone when I change the date to installation date.

commit 74f7b0d4 (2022-11-15): DateInterval: Fix sections with nil values

Closes #1471
closed 2022-11-15 by ledermann