solectrus/solectrus #1489: Fatal error when trying to enter the dashboard locally

flymann, 2022-11-22:
Hi Georg,

followed the in-house instructions and data is being pulled from the storage successfully but I got an error when entering the dashboard on http://localhost:3000 - there's the generic error screen only. 

Dev Console:
The request to http://localhost:3000/stats/now/inverter_power?field=inverter_power&period=now responds with a 500 :-(

It would be really great to get a hint ...

System: 
Docker version 20.10.11, build dea9396

[image]

Docker Application Logs:

```
senec-collector_1     | Successfully pushed record to InfluxDB
app_1                 | I, [2022-11-18T21:22:58.256468 #14]  INFO -- : [172.19.0.1] method=GET path=/stats/now/inverter_power format=html controller=StatsController action=index status=500 duration=19.90 view=0.00 user_agent=Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/106.0.0.0 Safari/537.36
app_1                 | F, [2022-11-18T21:22:58.257869 #14] FATAL -- : [172.19.0.1]   
app_1                 | [172.19.0.1] ActionView::Template::Error (NaN):
app_1                 | [172.19.0.1]      8:     - if calculator.present?
app_1                 | [172.19.0.1]      9:       div id="tab-#{period}" class="flex-auto w-full"
app_1                 | [172.19.0.1]     10:         - if period == 'now'
app_1                 | [172.19.0.1]     11:           = render DashboardNow::Component.new calculator:
app_1                 | [172.19.0.1]     12:         - else
app_1                 | [172.19.0.1]     13:           = render DashboardRange::Component.new calculator:,
app_1                 | [172.19.0.1]     14:                                                 period: permitted_params[:period],
app_1                 | [172.19.0.1]   
app_1                 | [172.19.0.1] app/services/scale.rb:13:in `round'
app_1                 | [172.19.0.1] app/services/scale.rb:13:in `result'
app_1                 | [172.19.0.1] app/components/balance_segment/component.rb:44:in `icon_size'
app_1                 | [172.19.0.1] app/components/balance_segment/component.html.slim:20:in `block in call'
app_1                 | [172.19.0.1] app/components/balance_segment/component.html.slim:1:in `call'
app_1                 | [172.19.0.1] app/components/balance/component.html.slim:5:in `block in call'
app_1                 | [172.19.0.1] app/components/balance/component.html.slim:4:in `each'
app_1                 | [172.19.0.1] app/components/balance/component.html.slim:4:in `call'
app_1                 | [172.19.0.1] app/components/dashboard_now/component.html.slim:2:in `call'
app_1                 | [172.19.0.1] app/views/stats/index.turbo_stream.slim:11
app_1                 | [172.19.0.1] app/controllers/stats_controller.rb:5:in `index'
app_1                 | E, [2022-11-18T21:22:58.259782 #14] ERROR -- honeybadger: ** [Honeybadger] Unable to send error report: API key is missing. id=f0e62d9e-e106-4429-9b0f-92e8c9bee71c level=3 pid=14
```

closed 2022-11-22 by ledermann

ledermann, 2022-11-22:
Thanks for reporting this error. I'm uncertain if I have fully understood the cause. It comes from the code which tries to scale the icon size by using the peak of the last 30 days. This should work even you don't have 30 days of data in your InfluxDB.
I have made a change to avoid NULL values, so maybe the next update fixes this.

If the error continues to occur on your side, you can try the development version of Solectrus by replacing the docker tag like this: Instead of `ghcr.io/solectrus/solectrus:latest` use `ghcr.io/solectrus/solectrus:develop` in your `docker-compose .yml`. 

But keep in mind that there are more unreleased changes included in the development version. The development version is the one which always runs on my personal instance (see https://solectrus.ledermann.dev)

Please tell me when the error is gone or not.

flymann, 2022-11-24:
Thanks a lot for investigation. Unfortunately there's still an error - the error message changed slightly.

```
app_1                 | F, [2022-11-24T21:49:40.874843 #32] FATAL -- : [172.20.0.1]
app_1                 | [172.20.0.1] ActionView::Template::Error (NaN):
app_1                 | [172.20.0.1]      8:     - if calculator.present?
app_1                 | [172.20.0.1]      9:       div id="tab-#{timeframe.id}" class="flex-auto w-full"
app_1                 | [172.20.0.1]     10:         - if timeframe.now?
app_1                 | [172.20.0.1]     11:           = render DashboardNow::Component.new calculator:
app_1                 | [172.20.0.1]     12:         - else
app_1                 | [172.20.0.1]     13:           = render DashboardRange::Component.new calculator:, timeframe:
app_1                 | [172.20.0.1]     14:
app_1                 | [172.20.0.1]
app_1                 | [172.20.0.1] app/services/scale.rb:13:in `round'
app_1                 | [172.20.0.1] app/services/scale.rb:13:in `result'
app_1                 | [172.20.0.1] app/components/balance_segment/component.rb:43:in `icon_size'
app_1                 | [172.20.0.1] app/components/balance_segment/component.html.slim:20:in `block in call'
app_1                 | [172.20.0.1] app/components/balance_segment/component.html.slim:1:in `call'
app_1                 | [172.20.0.1] app/components/balance/component.html.slim:5:in `block in call'
app_1                 | [172.20.0.1] app/components/balance/component.html.slim:4:in `each'
app_1                 | [172.20.0.1] app/components/balance/component.html.slim:4:in `call'
app_1                 | [172.20.0.1] app/components/dashboard_now/component.html.slim:2:in `call'
app_1                 | [172.20.0.1] app/views/stats/index.turbo_stream.slim:11
app_1                 | [172.20.0.1] app/controllers/stats_controller.rb:5:in `index'
app_1                 | I, [2022-11-24T21:49:40.881942 #32]  INFO -- : [172.20.0.1] method=GET path=/500 format=html controller=ErrorsController action=show status=500 duration=6.15 view=5.86 user_agent=Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/107.0.0.0 Safari/537.36
app_1                 | E, [2022-11-24T21:49:40.883575 #32] ERROR -- honeybadger: ** [Honeybadger] Unable to send error report: API key is missing. id=9e1ed492-1d05-4583-b44a-39f4287902d0 level=3 pid=32
```
Please ping me if I can provide helpful data for debugging

ledermann, 2022-11-25:
Ok, I just pushed another fix to the `develop` branch. If you like, you can try another update.

Background: The "NOW" tab resizes icons based on the current value in relation to the peak of the last 30 days (logarithmic). This calculation goes wrong on your side. With my latest change, any exception is caught and logged.

If there are still issues, I need some further information:

- Does your InfluxDB database contain measurements of the last 30 day? May be migrated data from [senec-importer](https://github.com/solectrus/senec-importer) or live data from [senec-collector](https://github.com/solectrus/senec-collector).
- Do the other tabs (day, week, month, year, all) fail, too? If not, please provide a full screenshot of the **month** tab (November 2022).
- Please provide another log snippet, there should be some additional debug information

flymann, 2022-11-25:
Awesome - I tried the new version and it works as expected now. 

So thank you again for this great support - I'm really happy to use this dashboard - you did a great job here !!

commit 01429901 (2022-11-22): Scale: Don't fail if there is no max value

Maybe fixes #1489