solectrus/solectrus #1496: Layout is defect when entering realtime tab

flymann, 2022-11-26:
For any reason the page layout is defect when entering the realtime tab.
Day / Week View is completely fine. 

Browser: Chrome Version 107.0.5304.121
Solectrus Release: 0.7.1 
branch: develop

No errors in DevTools / App Console stdout ...

Rendering Now Tab
[image]

### Rendering Day Tab
[image]

flymann, 2022-11-26:
I Checked the DOM and found a wrong value for font-size of the svg - when changing the value manually layout is fine again ...

**font-size: font-size: 8518% !!**
```
<div class="flex gap-2 lg:gap-3 flex-col items-center"><svg class="svg-inline--fa fa-bolt" style="font-size: 8518%;" aria-hidden="true" focusable="false" data-prefix="fas" data-icon="bolt" role="img" xmlns="http://www.w3.org/2000/svg" viewBox="0 0 448 512" data-fa-i2svg=""><path fill="currentColor" d="M349.4 44.6c5.9-13.7 1.5-29.7-10.6-38.5s-28.6-8-39.9 1.8l-256 224c-10 8.8-13.6 22.9-8.9 35.3S50.7 288 64 288H175.5L98.6 467.4c-5.9 13.7-1.5 29.7 10.6 38.5s28.6 8 39.9-1.8l256-224c10-8.8 13.6-22.9 8.9-35.3s-16.6-20.7-30-20.7H272.5L349.4 44.6z"></path></svg><!-- <i class="fa fa-bolt" style="font-size: 8518%;"></i> Font Awesome fontawesome.com --><div class="font-extralight uppercase tracking-wider leading-tight lg:block">Netzbezug</div><div><span><strong class="font-medium">0</strong><small>,2</small>&nbsp;<small>kW</small></span></div><template data-tippy-target="html"><div class="flex flex-col items-center justify-center"><i class="fa fa-bolt"></i>Netzbezug<br><span><strong class="font-medium">0</strong><small>,223</small>&nbsp;<small>kW</small></span></div></template></div>
```

flymann, 2022-11-26:
When checking the DOM I see some confusing values for height or font-size ...

[image]
[image]

ledermann, 2022-11-26:
Was the layout already wrong with `0.7.0` or only after installing `0.7.1`? I assume: Already wrong with `0.7.0`.

The problem seems to be related to #1489. For the "Now" tab, Solectrus calculates the peak values of all measurements to resize icons and height of flow animation accordingly. This goes wrong on your site (as seen, for example, in the uncommonly large font sizes). My bugfix in #1489 just suppresses an exception. There is still something wrong with your peak data.

Please check the logs for warnings while accessing the "Now" tab. Are there any?
Do you have real data for the last 30 days available in your database? Can you provide a screenshot of the "Month" tab?

Debugging is a bit hard because it seems you have some unexpected data in your InfluxDB database. I just added some _peak logging_ to the `develop` branch. Please, can you upgrade and then provide this log snippets? This would be very helpful. On my setup, these new log lines look like this:

```
I, [2022-11-26T14:50:23.440824 #11]  INFO -- : [...] Peak: {"bat_power_minus"=>3899, "bat_power_plus"=>2697, "grid_power_minus"=>5879, "grid_power_plus"=>23385, "house_power"=>6720, "inverter_power"=>8671, "wallbox_charge_power"=>20672}
```

flymann, 2022-11-26:
I checked out 0.7.1.1 - please find attached one debug log entry:
```
Successfully pushed record to InfluxDB
app_1                 | I, [2022-11-26T15:38:12.314783 #11]  INFO -- : [100.93.165.118] Peak: {"bat_power_minus"=>40, "bat_power_plus"=>0, "grid_power_minus"=>1, "grid_power_plus"=>19, "house_power"=>340, "inverter_power"=>297, "wallbox_charge_power"=>0}
```
There are no warnings in Chrome Devtools or app's stdout ...

And yes - at the moment I do have data for today only - will try to import the senec csv's this evening ...

[image]

ledermann, 2022-11-26:
Ok, thanks. Now I understand what happened: There is very little data, so the peak values are very low. And because the peak is cached for 24 hours (!) in Redis, it happens quickly that the current data is above the peak, so the calculation goes wrong - too large font size, broken layout etc.

I will build a fix soon. Thanks for helping!

flymann, 2022-11-26:
Great news :-) I imported the csv right now and have data for more than 30 days now.

After purging the redis dump & restarting the containers the Now View is fine again !! Unfortunately it's dark now and there's no battery power anymore :-( 

Thanks for support and a relaxing Erster Advent :-)

[image]

ledermann, 2022-11-26:
Fine! Meanwhile, I have pushed a fix so that values above the peak are no longer harmful. Peak logging is removed because it is no longer needed (and to not pollute the logs). 

There is no need to update your instance, as this is only relevant for small amounts of data.

And now - ready for Advent 🕯️ :)

closed 2022-11-26 by ledermann