solectrus/solectrus #1513: [Feature Request] Price calculation Car power consumption

ronzle, 2022-12-03:
First of all, very nice work, thanks for that !
The Senec app isn't very helpful, so your project is a huge benefit!

A nice feature would be, to add a cost calculation based on the power working costs for the power consumption of the car and put it on the overview pages as a dedicated circle. 

/ronny

ledermann, 2022-12-03:
Yes, that sounds interesting! Calculating the cost of charging the car in a certain time frame (day, week, month, year, total) would be a nice addition. I will think about it, thanks for this suggestion, @ronzle!

ronzle, 2023-01-04:
Hi 

do you plan a docker image as preview for 0.8 ? 
i could do some tests for you and give some feedback 

cheers 
Ronny

ledermann, 2023-01-04:
@ronzle Yes, there is a Docker image with this feature:
`ghcr.io/solectrus/solectrus:pr-1548`

But be warned: The implementation is still very rough. Both the calculation and the user interface will most likely change.

ledermann, 2023-03-07:
Recently, I improved and corrected the calculation and merged the code into `develop`, so the Docker image mentioned above no longer exists. You can use this for testing:
`ghcr.io/solectrus/solectrus:develop`

[image]

ledermann, 2023-03-12:
Today released v0.9.0 including this feature.