solectrus/solectrus #1520: wrong visualization on mobile phone and tablet

iot-sle, 2022-12-06:
There's a problem in showing values on mobile devices. It's the same on phone and tablet. I guess it's about calculating screen margins. Problem is independed of diagram type ( current, day, week, month...) and it happens at portrait and landscape mode.

At portrait mode:  the percentage values are missing the circle (1) and the chart (2) is overlapping to the right bar. There's a lot of space below the bars. I guess the chart should be placed there.
At landscape mode: the percentage circles are overlapping (1) as the chart (2) is. The value screen area is compressed. I guess the idea was to show the bars larger in vertical direction and the chart on the right hand.

Because it is the best overview I know, I want to use an older tablet for doing a continous visualization. I use the browers in full screen mode. So it would be a great idea to show all on one screen without scrolling. Without the problems described above it will work perfectly. Some space could be save if the line containing the period selector will be dropped and the selector will be shown in title bar between left edge and the menu icon. I don't know whether that's possible at all, and I'v no idea about the effort. 
[image]
[image]

ledermann, 2022-12-06:
It seems this is a browser issue.
The UI is tested on iPad and iOS using Safari 16. 

It seems you are using Android. You say "older tablet". Please, can you tell the exact browser version?
Can you try Solectrus on modern device and/or browser?

ledermann, 2022-12-16:
I just made a change to fix a layout issue on mid-sized displays. There was a bug when the display width was between 572 and 767 or between 884 and 1023.

@iot-sle The fix is available in the `develop` branch deployed on my personal instance (https://solectrus.ledermann.dev). Can you confirm that the error no longer occurs?

ledermann, 2022-12-19:
I assume that this bug is fixed