solectrus/solectrus #1611: Missing first day/month/year in overview

bennysweb, 2023-01-08:
In the overview for "Woche", "Monat" and "Jahr" is the first date missing. This is no bug especially in the actual version, I've observed this in the older versions too. Can this be improved?
Best regards,
Benny
[image]
[image]
[image]

ledermann, 2023-01-09:
@bennysweb Thank you for pointing this out. I will fix this soon.

closed 2023-01-09 by ledermann

ledermann, 2023-01-09:
Fixed in development branch (see online at https://solectrus.ledermann.dev).
Will be published in next regular release.

commit e7e015f4 (2023-01-09): Charts: Ensure to render first label on x-axis

Fixes #1611
commit c61a3d8c (2023-01-09): Charts: Ensure to render first label on x-axis

Fixes #1611