solectrus/solectrus #1675: dashboard does not communicate with influxdb

GuenniAC, 2023-01-30:
root@sau:/home/guenni/solectrus# docker ps -f health=unhealthy
CONTAINER ID   IMAGE                                COMMAND               CREATED          STATUS                      PORTS                                       NAMES
77574092c556   ghcr.io/solectrus/solectrus:latest   "docker/startup.sh"   33 minutes ago   Up 15 minutes (**unhealthy**)   0.0.0.0:3000->3000/tcp, :::3000->3000/tcp   solectrus-app-1

root@sau:/home/guenni/solectrus# docker restart 77574092c556
77574092c556

root@sau:/home/guenni/solectrus# docker ps -f health=unhealthy
CONTAINER ID   IMAGE                                COMMAND               CREATED             STATUS                      PORTS                                       NAMES
77574092c556   ghcr.io/solectrus/solectrus:latest   "docker/startup.sh"   About an hour ago   Up 43 minutes (**unhealthy**)   0.0.0.0:3000->3000/tcp, :::3000->3000/tcp   solectrus-app-1
root@sau:/home/guenni/solectrus# 

Auszug aus syslog:
root@sau:/var/log# tail -f syslog|grep -i docker
Jan 30 14:16:09 sau systemd[1]: run-docker-runtime\x2drunc-moby-77574092c556c1fee00117538a8de3f3690f10635b6f0d0a9546bbfccfdf4372-runc.B1PPXc.mount: Succeeded.
Jan 30 14:16:09 sau systemd[2671263]: run-docker-runtime\x2drunc-moby-77574092c556c1fee00117538a8de3f3690f10635b6f0d0a9546bbfccfdf4372-runc.B1PPXc.mount: Succeeded.
Jan 30 14:16:48 sau dockerd[1361846]: time="2023-01-30T14:16:48.088605434+01:00" level=info msg="Container failed to exit within 10s of signal 15 - using the force" container=77574092c556c1fee00117538a8de3f3690f10635b6f0d0a9546bbfccfdf4372
Jan 30 14:16:48 sau systemd[1]: docker-77574092c556c1fee00117538a8de3f3690f10635b6f0d0a9546bbfccfdf4372.scope: Succeeded.
Jan 30 14:16:48 sau systemd[1]: docker-77574092c556c1fee00117538a8de3f3690f10635b6f0d0a9546bbfccfdf4372.scope: Consumed 4.025s CPU time.
Jan 30 14:16:48 sau dockerd[1361846]: time="2023-01-30T14:16:48.172467131+01:00" level=info msg="ignoring event" container=77574092c556c1fee00117538a8de3f3690f10635b6f0d0a9546bbfccfdf4372 module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"
Jan 30 14:16:48 sau systemd[1783]: run-docker-netns-0a9ee033f7bf.mount: Succeeded.
Jan 30 14:16:48 sau systemd[2671263]: run-docker-netns-0a9ee033f7bf.mount: Succeeded.
Jan 30 14:16:48 sau systemd[1]: run-docker-netns-0a9ee033f7bf.mount: Succeeded.
Jan 30 14:16:48 sau systemd[1783]: run-docker-runtime\x2drunc-moby-77574092c556c1fee00117538a8de3f3690f10635b6f0d0a9546bbfccfdf4372-runc.TD48ap.mount: Succeeded.
Jan 30 14:16:48 sau systemd[2671263]: run-docker-runtime\x2drunc-moby-77574092c556c1fee00117538a8de3f3690f10635b6f0d0a9546bbfccfdf4372-runc.TD48ap.mount: Succeeded.
Jan 30 14:17:18 sau systemd[2671263]: run-docker-runtime\x2drunc-moby-77574092c556c1fee00117538a8de3f3690f10635b6f0d0a9546bbfccfdf4372-runc.yKpDXO.mount: Succeeded.
Jan 30 14:17:18 sau systemd[1]: run-docker-runtime\x2drunc-moby-77574092c556c1fee00117538a8de3f3690f10635b6f0d0a9546bbfccfdf4372-runc.yKpDXO.mount: Succeeded.
Jan 30 14:17:18 sau systemd[1783]: run-docker-runtime\x2drunc-moby-77574092c556c1fee00117538a8de3f3690f10635b6f0d0a9546bbfccfdf4372-runc.yKpDXO.mount: Succeeded.
Jan 30 14:17:48 sau systemd[2671263]: run-docker-runtime\x2drunc-moby-77574092c556c1fee00117538a8de3f3690f10635b6f0d0a9546bbfccfdf4372-runc.ALW6qm.mount: Succeeded.

Auf die Datenbank komme ich und der Collector läuft (Auszug der aktuellen Daten):  
#group,false,false,true,true,false,false,true,true
#datatype,string,long,dateTime:RFC3339,dateTime:RFC3339,dateTime:RFC3339,double,string,string
#default,mean,,,,,,,
,result,table,_start,_stop,_time,_value,_field,_measurement
,,0,2023-01-30T12:52:09.109513401Z,2023-01-30T13:52:09.109513401Z,2023-01-30T12:52:20Z,2.6,bat_charge_current,SENEC
,,0,2023-01-30T12:52:09.109513401Z,2023-01-30T13:52:09.109513401Z,2023-01-30T12:52:30Z,2.4,bat_charge_current,SENEC
,,0,2023-01-30T12:52:09.109513401Z,2023-01-30T13:52:09.109513401Z,2023-01-30T12:52:40Z,2.1,bat_charge_current,SENEC
,,0,2023-01-30T12:52:09.109513401Z,2023-01-30T13:52:09.109513401Z,2023-01-30T12:52:50Z,2.3,bat_charge_current,SENEC

_Originally posted by @GuenniAC in https://github.com/solectrus/senec-importer/issues/4#issuecomment-1408690869_

GuenniAC, 2023-01-30:
Seems that dashboard has a communication problem with the db:

`root@sau:/home/guenni/solectrus# docker ps
CONTAINER ID   IMAGE                                         COMMAND                  CREATED         STATUS                     PORTS                                       NAMES
0098f5251970   ghcr.io/solectrus/solectrus:latest            "docker/startup.sh"      8 minutes ago   Up 8 minutes (unhealthy)   0.0.0.0:3000->3000/tcp, :::3000->3000/tcp   solectrus-app-1

root@sau:/home/guenni/solectrus# docker logs --timestamps --details 0098f5251970|tail -10
2023-01-30T17:22:28.302816755Z  Waiting for InfluxDB on sau:8086 ...
2023-01-30T17:22:29.306898335Z  Waiting for InfluxDB on sau:8086 ...
2023-01-30T17:22:30.310783738Z  Waiting for InfluxDB on sau:8086 ...
2023-01-30T17:22:31.315017100Z  Waiting for InfluxDB on sau:8086 ...
2023-01-30T17:22:32.319099525Z  Waiting for InfluxDB on sau:8086 ...
2023-01-30T17:22:33.323352416Z  Waiting for InfluxDB on sau:8086 ...
2023-01-30T17:22:34.327676518Z  Waiting for InfluxDB on sau:8086 ...
2023-01-30T17:22:35.331951602Z  Waiting for InfluxDB on sau:8086 ...
2023-01-30T17:22:36.336113065Z  Waiting for InfluxDB on sau:8086 ...
2023-01-30T17:22:37.340461061Z  Waiting for InfluxDB on sau:8086 ...
root@sau:/home/guenni/solectrus# 

`

ledermann, 2023-01-30:
Ich denke, wir haben hier ein Netzwerkproblem: Der Dashboard-Container versucht, die Influx-Datenbank über `sau:8086` zu erreichen (`sau` ist wohl der Hostname), richtig wäre aber `influxdb`, also der Service-Name der Influx-Datenbank in der `docker-compose.yml`. 

Gesteuert wird das über die Umgebungsvariable `INFLUX_HOST`, siehe [Beispiel](https://github.com/solectrus/hosting/blob/main/guide/raspberry-pi/.env).

Vermutlich hast Du beim Importieren der CSV-Dateien die Umgebungsvariablen geändert.

GuenniAC, 2023-01-30:
Korrekt. Das Dashboard läuft wieder.
Besten Dank!

closed 2023-01-31 by GuenniAC