solectrus/solectrus #1680: Dashboard zeigt uneinheitliche Statistiken

GuenniAC, 2023-01-31:
Das importieren der Statistiken von Mai 22 bis zum 26.1. (dem Tage der Inbetriebnahme) hat funktioniert, jedoch ist die Darstellung im Dashboard "uneinheitlich".
- Monatsstatistik:  ich sehe im Januar alle Tage mit Daten, vor dem 1.1. aber nichts
- Wochenstatistik: keine Daten für KW1-3, KW4: nichts vor dem 26.1.
- Jahresstatistik: keine Daten für das Jahr 2022

Ein Restart der Services hat nichts gebracht. Hier ein paar Screenshots:

[image]
[image]
[image]

ledermann, 2023-01-31:
Ich denke, das liegt am Caching. SOLECTRUS speichert die Antworten auf bestimmte Influx-Queries in Redis, was bei wiederholten Abfragen ein großer Performance-Gewinn ist. Dabei wird davon ausgegangen, dass sich die Messwerte von **in der Vergangenheit** liegenden Zeiträume niemals ändern können. Das ist im Normalbetrieb auch der Fall, aber eben nicht, wenn man nachträglich den [SENEC-Importer](https://github.com/solectrus/senec-importer) verwendet hat.

Langer Rede kurzer Sinn: Du solltest den Redis-Cache leeren. Am einfachsten wie folgt:

```bash
docker exec -it solectrus-redis-1 redis-cli FLUSHALL
``` 

Der Befehl geht davon aus, dass der Redis-Container `solectrus-redis-1` heißt. Falls er anders heißen sollte, bitte vorher anpassen. Oder, falls das nicht klappt, alles herunterfahren (`docker compose down`), den Redis-Ordner manuell leeren und dann wieder alles neu starten (`docker compose up`).

GuenniAC, 2023-01-31:
Das hat nur teilweise funktioniert.
Die Wochenstatistik für KW4 ist vollständig, der Rest ist unverändert.
Auch nach dem runterfahren, manuellem Löschen von redis/dump.rdb und Wiederstart.

ledermann, 2023-01-31:
Prüf mal die Umgebungsvariable `INSTALLATION_DATE`. Es werden nämlich grundsätzlich keine Daten angezeigt, die **vor** diesem Datum liegen.

GuenniAC, 2023-01-31:
Ja, richtig. Die stand auf 23.01.23. Nach restart (und löschen von redis/dump.rdb) gibt es nun Serverfehler 500, wenn ich auf KW3 oder andere frühere Zeiträume gehe.

ledermann, 2023-01-31:
Oh je... Check mal die erfassten Preise (unter `/settings/prices`) und stelle sicher, dass Strompreise und Einspeisevergütungen ab dem eingestellten `INSTALLATION_DATE` vorhanden sind.

GuenniAC, 2023-01-31:
Preise sind:
ELECTRICITY_PRICE=0.28
FEED_IN_TARIFF=0.07

Habe aber einen Hinweis im Redis-log gefunden:
root@sau:/home/guenni/solectrus# docker logs --timestamps --details solectrus-redis-1
2023-01-31T09:02:06.111874060Z 1:C 31 Jan 2023 09:02:06.110 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
2023-01-31T09:02:06.111910024Z 1:C 31 Jan 2023 09:02:06.110 # Redis version=7.0.8, bits=64, commit=00000000, modified=0, pid=1, just started
2023-01-31T09:02:06.111914378Z 1:C 31 Jan 2023 09:02:06.110 # Warning: no config file specified, using the default config. In order to specify a config file use redis-server /path/to/redis.conf
2023-01-31T09:02:06.111917690Z 1:M 31 Jan 2023 09:02:06.111 * monotonic clock: POSIX clock_gettime
2023-01-31T09:02:06.113505680Z 1:M 31 Jan 2023 09:02:06.112 * Running mode=standalone, port=6379.
2023-01-31T09:02:06.113525807Z 1:M 31 Jan 2023 09:02:06.112 # Server initialized
2023-01-31T09:02:06.113529402Z 1:M 31 Jan 2023 09:02:06.112 # WARNING Memory overcommit must be enabled! Without it, a background save or replication may fail under low memory condition. Being disabled, it can can also cause failures without low memory condition, see https://github.com/jemalloc/jemalloc/issues/1328. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
2023-01-31T09:02:06.113543287Z 1:M 31 Jan 2023 09:02:06.112 * Ready to accept connections

GuenniAC, 2023-01-31:
OK, das war natürlich auch auf den 23.1.23 gesetzt. Jetzt sehe ich alles.
Nun werde ich mal weiter prüfen.
Vielen Dank für die Unterstützung!

closed 2023-01-31 by GuenniAC

ledermann, 2023-01-31:
Die Preise in den Umgebungsvariablen werden beim **ersten** Start von SOLECTRUS in die Postgres-Datenbank geschrieben, gültig ab `$INSTALLATION_DATE`. Anschließend sind nur noch die Preise in der Datenbank relevant. Im UI (unter `/settings/prices/`) können Preise editiert werden, damit man auch Preisänderungen im Zeitverlauf verwalten kann. Jeder Preis hat also ein Datum, ab dem es gültig ist.

Bei Dir gibt es nun vermutlich keine Preise für die Zeiträume, die Du nachträglich importiert hast. Daher bitte im Browser unter `/settings/prices` sicherstellen, dass ab `$INSTALLATION_DATE` Preise vorhanden sind, also bei Dir ab Tag x in 2022.

Noch ein Hinweis zu Deinem ersten Kommentar: Mit "Tage der Inbetriebnahme" meint SOLECTRUS das Datum, an dem die Photovoltaikanlage auf dem Dach installiert wurde und erstmalig Strom produziert hat. Also nicht das Datum der Installation von SOLECTRUS :-)

Die Warnung im Redis-Log kann übrigens ignoriert werden.

GuenniAC, 2023-01-31:
Hätte ich den Kommentar im .env-File aufmerksam gelesen hätte ich es gewusst;-) Oder ich habe beim setzen dieser Variable noch nichts vomn der Möglichkeit des nachimportierens gewusst.
Als nächstes werde ich wohl dann mal nach einem Backup schauen. Hast du ja auch im Angebot!
Nochmals danke!