solectrus/solectrus #1693: Show Senec Status

iot-sle, 2023-02-03:
It would be nice to see the status of Senec Home near to the "live"/"fail" symbol in header. Or perhaps as a mouse hover overlay when "live" symbol is shown.
 STATISTIC:   {  CURRENT_STATE: 15 }

some  values can be found here: https://gist.github.com/smashnet/82ad0b9d7f0ba2e5098e6649ba08f88a

ledermann, 2023-02-04:
This is a great idea! I have successfully implemented it:

- The state names are obtained from the source file located at `http://[senec-ip]/js/DE-de.js`. This JS file is parsed when the senec-collector starts, ensuring that names and mappings are always up-to-date. 
- The senec-collector retrieves the state from the SENEC device and pushes it, along with all other measurements, to InfluxDB. 
- The dashboard displays the current state in the header instead of the static text "LIVE". Maybe this requires some more tweaks.

A demo can be viewed at https://solectrus.ledermann.dev. This feature will be included in the next official release.

ledermann, 2023-02-05:
Just released v0.6.0 of the senec-collector, which is required to see the status in the UI.

iot-sle, 2023-02-06:
Thank you. Nice implementation of this new feature.

ledermann, 2023-02-17:
Released in [v0.8.0](https://github.com/solectrus/solectrus/releases/tag/v0.8.0)

closed 2023-02-17 by ledermann