solectrus/solectrus #1726: Autarky values  missing in diagram

iot-sle, 2023-02-13:
For some reason the autarky diagram isn't filled in all the time. I attached 2 screenshots. It seems that the values are available, because the sum is shown. If you need more details or access to the dbs, pls send a personal mail.
[Screenshot_20230213-134510_Opera](https://user-images.githubusercontent.com/74416643/218515857-d9ab04b2-3692-43dd-bd5a-c5605d2d1159.jpg)
[Screenshot_20230213-134414_Opera](https://user-images.githubusercontent.com/74416643/218515880-d1bd9389-3d62-4f6b-af3e-b212f3baed22.jpg)

ledermann, 2023-02-13:
There are two things I suggest to check:

1) Is your ENV var `INSTALLATION_DATE`correct? It should point to the installation date of your SENEC (= date with first data)

2) Maybe there is a caching issue. You can reset your Redis cache with this command:

    `docker exec -it [solectrus-redis-container-name] redis-cli FLUSHALL`
    e.g. `docker exec -it solectrus-redis-1 redis-cli FLUSHALL`

   This may happen if you have imported data (using [senec-importer](https://github.com/solectrus/senec-importer)) **after** the app was already used and has filled the cache.

If neither of these helps: What about other charts? Are they also empty? What about `house_power`chart in August 2022?

iot-sle, 2023-02-15:
no success by executing the flushall. All other charts are fine.

ledermann, 2023-02-16:
Strange. Are there any errors in the browser console or in the container logs?

Is your SOLECTRUS installation accessible to the public? May I have a look at it? Or can you give me private access (via e-mail)?

iot-sle, 2023-02-17:
Did you receive my email about accessing my instalation? I just updated to 0.8.0 and it's still the same behavior according this chart. I like the live-status display.

ledermann, 2023-02-17:
No email so far. Please send to info@solectrus.de

ledermann, 2023-02-18:
Thanks to @iot-sle, this error was identified and confirmed. It occurs for the periods that were filled with SENEC's CSV data via the [senec-importer](https://github.com/solectrus/senec-importer). These data sets do not contain any measurements for the wallbox, meaning that these values are stored as `NULL`, which breaks the autarky calculation by InfluxDB.

A fix has been implemented - NULL values are now converted to zero (`0.0`). The fix will be included in the next update (expected to be v0.8.1).

closed 2023-02-18 by ledermann