solectrus/solectrus #1733: Display values on the dashboard also in watts in real time instead of kwh

igriepp, 2023-02-15:
Hello, would it be possible to display the values in watts instead of kwh on the dashboard? In the Senec app you have the option to choose between kwh and watts. king regards IGriepp

igriepp, 2023-02-15:
Done

ledermann, 2023-02-15:
Done? Not yet :-) 

SOLECTRUS displays smaller values (less than 1000 kWh or kW) as **kWh**, larger ones (greater than 1000 kWh) as **MWh**. Currently, there is no option to customize this behavior. 

Perhaps I will consider such an option. Or maybe it's better to display tiny values (less than 1 kWh) as Watt. Will see.

igriepp, 2023-02-15:
Thank you for your reply. I think that the display in watts is more accurate than kw. I would be happy if a conversion or selection between watts and kw on the dashboard would be possible in the future. Many greetings IGriepp