solectrus/solectrus #1734: [Feature Request] Display Current Battery Precentage

Twooearly, 2023-02-15:
As I saw, the Current Battery Percentage is shown on the new Layout there:
[image]

Also this Value ist monitored in the InfluxDB and can be accessed at `https://solectrus-domain.com/bat_fuel_charge/`
Could you implement it, that the chart at this URL gets a proper Color and can be accessed when clicking on the Current Battery Percentage.

I hope this isn't too hard and it would be a huge improvement to daily use for me.
Hope you can make use of this. Have a great Day!

ledermann, 2023-02-16:
Well observed :) Yes, there is a hidden battery chart, which works fine for line charts of "NOW" and "DAY" (currently not colorized). Example here:
https://solectrus.ledermann.dev/bat_fuel_charge/2023-02-16

However, it does not make sense for bar charts (Week, Month, Year, All). This is the reason I have not yet added a link to this chart. I still need to figure out the best way to solve this.

closed 2023-04-01 by ledermann

ledermann, 2023-04-01:
Just released as v0.10.0

PR #1849 (2023-04-01): Add diagrams for bat_fuel_charge and case_temp
Resolves #1734 
Resolves #1054