solectrus/solectrus #1743: Error 500 beim Aufruf des Dashboards

sebastianmaisel, 2023-02-18:
Hi,
wir wollten heute auf einem Synology Server Solectrus installieren. 
Leider bleiben wir beim Versuch, das Dashboard zu öffnen, beim Fehler 500 (Internal Server Error) hängen.
Als Fehlermeldung in Docker-Compose erhalten wir folgenden Fehler: 

```
app_1 | F, [2023-02-18T12:51:58.383197 #12] FATAL -- : [172.19.0.1]
app_1 | [172.19.0.1] ActionView::Template::Error (NaN):
app_1 | [172.19.0.1] 6: turbo-stream action="morph" target="stats" children-only= true
app_1 | [172.19.0.1] 7: template
app_1 | [172.19.0.1] 8: - if timeframe.now?
app_1 | [172.19.0.1] 9: = render DashboardNow::Component.new calculator:
app_1 | [172.19.0.1] 10: - else
app_1 | [172.19.0.1] 11: = render DashboardRange::Component.new calculator:, timeframe:
app_1 | [172.19.0.1] 12:
app_1 | [172.19.0.1]
app_1 | [172.19.0.1] app/services/scale.rb:17:in `round'
app_1 | [172.19.0.1] app/services/scale.rb:17:in `result'
app_1 | [172.19.0.1] app/components/balance_segment/component.rb:46:in `icon_size'
app_1 | [172.19.0.1] app/components/balance_segment/component.html.slim:21:in `block in call'
app_1 | [172.19.0.1] app/components/balance_segment/component.html.slim:1:in `call'
app_1 | [172.19.0.1] app/components/balance/component.html.slim:5:in `block in call'
app_1 | [172.19.0.1] app/components/balance/component.html.slim:4:in `each'
app_1 | [172.19.0.1] app/components/balance/component.html.slim:4:in `call'
app_1 | [172.19.0.1] app/components/dashboard_now/component.html.slim:32:in `call'
app_1 | [172.19.0.1] app/views/stats/index.turbo_stream.slim:9
app_1 | [172.19.0.1] app/controllers/stats_controller.rb:5:in `index'
```

Habe gesehen, dass ein anderer Nutzer in #1489 bereits dasselbe Problem mit anderer Ursache hat.
Der Senec-Collector scheint Daten erfolgreich zu empfangen, auch die Datenbankkommunikation läuft laut compose-logs einwandfrei. Irgendeine Idee, wo ich am Debugging ansetzen kann?

Vielen Dank im Voraus!

ledermann, 2023-02-18:
Hm, ein Fehler 500 ist definitiv etwas, das ich beheben möchte. Ich habe aber noch keine Idee, was hier die Ursache sein könnte. Es hängt vermutlich damit zusammen, dass noch zu wenig oder gar keine Daten vorhanden sind.

Grundsätzlich versucht SOLECTRUS, die Darstellung auf die Leistung der PV-Anlage abzustimmen, d.h. die Größe von Icons und der Fluss-Animation orientiert sich am Peak-Wert der letzten 30 Tage. Da geht wohl noch etwas schief, was ich hier bei mir aber noch nicht reproduzieren kann.

Ich gehe davon aus, dass Du die aktuelle Version v0.8.0 installiert hast. Versuchte bitte mal manuell diese URLs aufzurufen:

- /inverter_power/2023-02-18
- /top10/day/inverter_power
- /settings

Ich vermute, die funktionieren alle. Der Fehler wird nur beim Tab "VERLAUF" auftreten und dort auch nur bei "JETZT".

Was Du versuchen kannst:

1) Redis-Cache löschen: 

  `docker exec -it [solectrus-redis-container-name] redis-cli FLUSHALL`
  Also z.B. `docker exec -it solectrus-redis-1 redis-cli FLUSHALL``

2) Warten :-) Wenn der SENEC-Collector erfolgreich Daten in die InfluxDB schreibt, sollte sich das Problem spätestens nach 24h von selbst erledigt haben.

Dennoch: Fehler dieser Art sollten niemals auftreten und ich bin sehr daran interessiert, diesen zu beheben. Falls Du noch irgendwelche Beobachtungen machst, dann würden mich die sehr interessieren.

ledermann, 2023-02-18:
Ich habe für die Fehlersuche die betreffende Code-Stelle um zusätzliches Logging ergänzt. 

Wenn Du magst, probiere doch bitte das `develop`-Image aus (`ghcr.io/solectrus/solectrus:develop`). Das Logfile (vom Start des Containers bis zum ersten Aufruf der App im Browser) wäre interessant.

sebastianmaisel, 2023-02-18:
Danke dir für die schnelle Rückmeldung.
Habe die oben genannten Vorschläge alle mal durchgetestet: 

Bzgl. der drei URLs: Bis auf die /settings, die ein 404 zurückwirft, funktioniert alles- du hast also Recht, dass es an dem "Jetzt" liegt :-). Habe vorhin mal sämtliche Versionen der App getestet (bis zur v0.7 zurück), der Fehler ist überall gleich geblieben...

1. Cache ist gelöscht, Besserung hat das glaube ich nicht gebracht
2. Das mit dem 24h-Warten mach ich definitiv mal, bin gespannt ob es alles klappt - leider werden aktuell Arbeiten an der Anlage gemacht, deswegen können keine Daten für die Einspeisung generiert werden.

Das Develop-Image läuft jetzt, anbei mal die Logs von Starten des Docker-Compose bis zum Aufruf der Root-Page:

`app_1                 | Starting SOLECTRUS...
app_1                 | Version: v0.8.0-17-g0d6e86e - 2023-02-18T09:11:31+01:00
app_1                 | ----------------
forecast-collector_1  | Forecast collector for SOLECTRUS, Version 0.2.0, built at 2023-02-07T09:08:10.238Z
forecast-collector_1  | https://github.com/solectrus/forecast-collector
forecast-collector_1  | Copyright (c) 2020,2023 Georg Ledermann, released under the MIT License
forecast-collector_1  |
db_1                  |
db_1                  | PostgreSQL Database directory appears to contain a database; Skipping initialization
db_1                  |
app_1                 | InfluxDB is up and running!
app_1                 | influxdb (172.19.0.2:8086) open
influxdb_1            | Command "print-config" is deprecated, use the influx-cli command server-config to display the configuration values from the running server
forecast-collector_1  | Using Ruby 3.2.0 on platform x86_64-linux-musl
forecast-collector_1  | Pulling from api.forecast.solar every 900 seconds
forecast-collector_1  | Pushing to InfluxDB at http://influxdb:8086, bucket my-solectrus-bucket
forecast-collector_1  |
influxdb_1            | Command "print-config" is deprecated, use the influx-cli command server-config to display the configuration values from the running server
db_1                  | 2023-02-18 20:30:23.296 UTC [1] LOG:  starting PostgreSQL 15.2 on x86_64-pc-linux-musl, compiled by gcc (Alpine 12.2.1_git20220924-r4) 12.2.1 20220924, 64-bit
db_1                  | 2023-02-18 20:30:23.296 UTC [1] LOG:  listening on IPv4 address "0.0.0.0", port 5432
db_1                  | 2023-02-18 20:30:23.296 UTC [1] LOG:  listening on IPv6 address "::", port 5432
senec-collector_1     | SENEC collector for SOLECTRUS, Version 0.6.0, built at 2023-02-05T11:44:03.299Z
senec-collector_1     | https://github.com/solectrus/senec-collector
senec-collector_1     | Copyright (c) 2020,2023 Georg Ledermann, released under the MIT License
senec-collector_1     |
redis_1               | 1:C 18 Feb 2023 20:30:23.937 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
redis_1               | 1:C 18 Feb 2023 20:30:23.937 # Redis version=7.0.8, bits=64, commit=00000000, modified=0, pid=1, just started
redis_1               | 1:C 18 Feb 2023 20:30:23.937 # Warning: no config file specified, using the default config. In order to specify a config file use redis-server /path/to/redis.conf
app_1                 | db (172.19.0.3:5432) open
senec-collector_1     | Using Ruby 3.2.0 on platform x86_64-linux-musl
senec-collector_1     | Pulling from SENEC at 10.0.5.187 every 5 seconds
senec-collector_1     | Pushing to InfluxDB at http://influxdb:8086, bucket my-solectrus-bucket
senec-collector_1     |
senec-collector_1     | Getting state names from SENEC by parsing source code...
redis_1               | 1:M 18 Feb 2023 20:30:23.938 * monotonic clock: POSIX clock_gettime
influxdb_1            | Command "print-config" is deprecated, use the influx-cli command server-config to display the configuration values from the running server
influxdb_1            | 2023-02-18T20:30:22.	info	found existing boltdb file, skipping setup wrapper	{"system": "docker", "bolt_path": "/var/lib/influxdb2/influxd.bolt"}
senec-collector_1     | OK, got 99 state names
redis_1               | 1:M 18 Feb 2023 20:30:23.939 * Running mode=standalone, port=6379.
app_1                 | PostgreSQL is up and running!
app_1                 | Preparing database...
influxdb_1            | Command "print-config" is deprecated, use the influx-cli command server-config to display the configuration values from the running server
db_1                  | 2023-02-18 20:30:23.449 UTC [1] LOG:  listening on Unix socket "/var/run/postgresql/.s.PGSQL.5432"
senec-collector_1     |
senec-collector_1     | Got record #1 from SENEC at 10.0.5.187: AKKU LEER, Inverter 0 W, House 1266 W, 2023-02-18 20:30:27 +0000
influxdb_1            | Command "print-config" is deprecated, use the influx-cli command server-config to display the configuration values from the running server
forecast-collector_1  | #1: Getting data from https://api.forecast.solar/estimate/50.12345/6.12345/30/20/8.4?time=seconds ... OK
influxdb_1            | Command "print-config" is deprecated, use the influx-cli command server-config to display the configuration values from the running server
redis_1               | 1:M 18 Feb 2023 20:30:23.939 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
redis_1               | 1:M 18 Feb 2023 20:30:23.939 # Server initialized
[... cut ...] lla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/16.3 Safari/605.1.15
app_1                 | ** [Honeybadger] Unable to send error report: API key is missing. id=4451e141-d139-483f-914e-5b8a1f5b9baa level=3 pid=11
senec-collector_1     |
senec-collector_1     | Got record #39 from SENEC at 10.0.5.187: AKKU LEER, Inverter 0 W, House 1277 W, 2023-02-18 20:33:38 +0000
senec-collector_1     | Successfully pushed record to InfluxDB
senec-collector_1     |`

ledermann, 2023-02-19:
Ok, besten Dank für die Logs! Ich habe jetzt einen ersten Fix erstellt, der das Problem beheben sollte - neues Development-Image liegt vor, daher bitte `docker pull ghcr.io/solectrus/solectrus:develop` und neu starten.

Falls Du danach in den App-Logs ein "WARNING: Invalid input" sehen solltest, teile mir das bitte mit.

Ergänzend noch:

- Die Sache mit den 24h liegt daran, dass SOLECTRUS regelmäßig den Peak der Messwerte aus den letzten 30 Tagen zu ermitteln versucht und davon abhängig die Visualisierung anpasst. Dieser Peak wird für 24h im Cache gespeichert. Wenn es aber gar keine oder nur wenig Daten aus den letzten 30 Tagen gibt, klappt das nicht gut. Dennoch sollte es natürlich nie zu einem Fehler kommen.
- Der von mir angegebene Settings-Link war falsch, korrekt ist `/settings/prices`

sebastianmaisel, 2023-02-19:
Der Fehler ist durch deinen neuesten Fix nun behoben! :-)
Werde die Funktion mal die nächsten Tage mal beobachten, bislang hab ich kein "Invalid Input" in den Logs entdeckt!

Tausend Dank für die schnelle Reaktion - bin sehr gespannt, wie sich das Projekt hier so weiter entwickelt und bin jetzt schon ganz begeistert!

ledermann, 2023-02-19:
Super, freut mich. Ich habe daher nun die v0.8.1 released, Du kannst somit auch wieder auf das latest-Image wechseln.

closed 2023-02-19 by ledermann