solectrus/solectrus #1750: [Feature Request] Flop 10

spirit0607, 2023-02-20:
Hallo,
in Ergänzung zu der Top 10 Funktion, wäre eine Übersicht der Flop 10 toll! So könnte man z.B. verschiedene Wintermonate der vergangenen Jahre vergleichen. Diese tauchen ja verständlicherweise nie in den Top 10 auf. Vielleicht lässt sich das realisieren.

Vielen Dank nochmal für dieses tolle Projekt!!
Viele Grüße 
Julian

ledermann, 2023-02-21:
Ja, gute Idee - ist notiert :-)

Zeppelin-PV, 2023-02-26:
hallo
genau so wie #1756 ist auch dies eine gute Idee so wie von Julian vorgeschlagen
... für diese großartige Alternative zu Senec viel Lob an ledermann
LG

closed 2023-03-05 by ledermann

ledermann, 2023-03-12:
v0.9.0 habe ich vorhin veröffentlicht. Die Top 10 kann nun auf- und absteigend sortiert werden. Den Begriff "Flop 10" wollte ich vermeiden :-)

Hier ein Beispiel mit den schlechtesten Monaten meiner PV-Anlage:
https://solectrus.ledermann.dev/top10/month/inverter_power/asc

Detail am Rande: Bei aufsteigender Sortierung werden sowohl der Installationsmonat als auch der aktuelle Monat nicht berücksichtigt, weil das die Darstellung verzerren würde.

ledermann, 2023-03-14:
Interessant, in der Tages-Anzeige taucht der 20.06.2022, also zwei Tage **vor** dem von Dir genannten Installationsdatum.  Und die Monatsanzeige enthält den Juni 2022, was nicht sein soll.

@spirit0607 Bitte prüfe doch mal, ob bei Deiner Installation wirklich das Installationsdatum korrekt angegeben ist. Dieses findet sich nicht in der Benutzeroberfläche, sondern in der .env-Datei von Docker in der Variablen `INSTALLATION_DATE`. Ich vermute, dass da etwas anderes drinsteht.

spirit0607, 2023-03-14:
Ahh Tippfehler! 
Das sollte nicht 22.06. heißen, sondern 20.06.! Also das passt schon. Habe gerade auch in die .env Datei geschaut. Da steht auch der 20.06.22 drin (INSTALLATION_DATE=2022-06-20)

ledermann, 2023-03-15:
Sehr seltsam. Mit dieser Einstellung sollte es funktionieren. Du bist Dir auch sicher, dass Du in die richtige .env-Datei geguckt hast, diese auch von docker-compose verwendet wird und der Container nach einer Änderung auch neu gestartet wurde? Du kannst die aktiven ENV-Variablen des laufenden Containers wie folgt anzeigen lassen:

```bash
docker exec solectrus-app-1 /usr/bin/env
```

(Container-Name ggfs. anpassen!, siehe `docker ps`)

Bitte prüfe, ob da wirklich `INSTALLATION_DATE=2022-06-20` ausgegeben wird.

Generell: Ja, es ist nicht ideal, solche Einstellungen per ENV machen zu müssen. Kommt Zeit, kommt UI :-)

bennysweb, 2023-03-15:
Dazu hätte ich noch eine Anmerkung/Ergänzung: Die vollständige Periode sollte bei der Jahresansicht nicht gelten. Meine Anlage ging letztes Jahr im März in Betrieb, damit ist 2022 keine vollständige Periode und somit wird in der Flop10 Jahresansicht leider nichts angezeigt. 
[image]

ledermann, 2023-03-16:
@bennysweb Stimmt, danke für den Hinweis, ich korrigiere das (in #1828)

@spirit0607 Falls es weiterhin Probleme oder Unstimmigkeiten gibt, bitte ein neues Issue anlegen, sonst kommen wir hier durcheinander :-)