solectrus/solectrus #1756: Add more parameters and second roof

jpp2023, 2023-02-24:
Hi, 
Great tool. I am running a system that is distributed over two parts of the roof with different orientation. Now I am interested in evaluating the performance of the  individual sections. I see in the Webserver of the SENEC the different MPP powerlevels[U,I,P] can be read out and I know how my strings are connected so could map that to evaluate then performance of the different sections. 
Question is could you include MPP1, MPP2, MPP3 parameters to the database and visualization? And could a second roof with different kWp and orientation for the predicted power be included?
Thanks, JP

ledermann, 2023-02-24:
Thank you for the suggestion!

The individual performance values for MPP 1/2/3 could be extracted and written to the InfluxDB. This would be a relatively simple step. However, finding a good and easily understandable visualization is its own challenge.

I'll leave this issue open. Since my own PV system only operates a single side of the roof, I don't have any personal benefit from such a feature. So, it probably won't be implemented anytime soon - unless someone else wants to take care of it (hey, it's an open-source project!) or maybe someone wants to sponsor me for this feature :-)

Zeppelin-PV, 2023-02-26:
hallo
die Anzeige der drei MPP's und somit die Erzeugung der Strings wäre wirklich vorteilhaft
könnte mir vorstellen diese als drei Balken innerhalb der Erzeugung anzuzeigen, farblich leicht abgesetzt
sinnvoll vielleicht erst ab Gesamterzeugung <1kW da sonst kein Platz dafür da ist
oder als eigene Grafik
... ansonsten großartige Alternative zu Senec
LG

noffycws, 2023-03-28:
Die Aufteilung der einzelnen Strings wäre schon eine Idee. Somit sieht man, wie die einzelnen Strings auch Strom erzeugen. Ich selbst habe 3 Strings, wobei ein String auf dem Gartendach ist und 2 Strings auf Straßenseite. SENEC liefert ja die Daten. Welche Leistung welcher String maximal liefert, weiß man ja bzw. kann man berechnen.

Bzgl. Darstellung mal, wie ich es derzeit über Grafana mache:

[image]

Man sieht deutlich die grüne Linie, die aufgrund des anderen Ausrichtung des Hausdaches früher Strom erzeugt. Die beiden anderen Linien zeigen die Strings zur Straßenseite. Vielleicht kann man das als Detail-Sicht in SOLECTRUS abbilden. Dass man zunächst die Gesamtleistung sieht (wie jetzt auch) und durch einen weiteren Klick man sich die einzelnen Strings anzeigen lassen kann.

StevieC121176, 2023-08-11:
> hallo die Anzeige der drei MPP's und somit die Erzeugung der Strings wäre wirklich vorteilhaft könnte mir vorstellen diese als drei Balken innerhalb der Erzeugung anzuzeigen, farblich leicht abgesetzt sinnvoll vielleicht erst ab Gesamterzeugung <

da ich die Idee gut finde und 2 PV Anlagen habe (und damit an einer Umsetzung sehr interessiert bin), hier mal ein grafischer Entwurf wie das aussehen könnte. (Farben im Diagramm sind nicht angepasst).

 
[image]

ledermann, 2023-08-13:
Besten Dank für den Input, @StevieC121176 - das sieht hervorragend aus 👍

GalOR7, 2024-10-18:
Ja wäre echt super. Frage ja auch schon länger danach 😇

StevieC121176, 2024-12-03:
Ich finde deine neue Ansicht für die Wärmepumpe und die Verbraucher äußerst gut gelungen. Das bringt mich auf die Idee, da das Grundgerüst da ist, dieses auch für mehrere Dachflächen umzusetzen?
Das würde sich eigentlich für dieses Szenario anbieten.
Ich habe z.Bsp 2 Dachflächen aber 5 MPP Tracker, da einige Teile über den Tag auch mal verschattet sind. In so einer Extra Ansicht könnte man das schön darstellen und hätte einen super Überblick über die einzelnen Erträge.
Wäre das evtl. machbar?

[image]
[image]
[image]

ledermann, 2024-12-03:
@StevieC121176 Ja, gute Idee, das wäre denkbar. Derzeitig konzentriere ich mich aber darauf, die neuen Darstellungen für Verbraucher und Wärmepumpe fertig zu bekommen, da ist noch einiges zu tun...

ledermann, 2024-12-09:
@GalOR7 Momentan gibt es noch zu wenig zahlende User, der Beitrag von Einzelnen zur Finanzierung eines konkreten Features wäre daher viel zu hoch, damit es sich gleichzeitig für mich rechnen würde.

Ich hatte schon mal überlegt, an jedes offene Issue ein Preisschild zu kleben, aber dann würde da jeweils ein niedriger bis mittlerer vierstelliger Betrag stehen (um den echten Aufwand abzudecken). Wenn sich das auf wenige Interessenten verteilt, würde ich die Privatnutzer nur vergraulen.

Falls jemand Alternativvorschläge zur Finanzierung hat, kann mir gerne eine Mail schreiben :)

GalOR7, 2024-12-09:
Ja verstehe und so wie du das schreibst hast du locker 100k Euro in dem Projekt stecken wenn du alles nach normalen Sätzen als Developer abrechnen könntest. Da steckt richtig "Herzblut" drin in Solectrus. Aber das mit den Preisschildern ist keine schlechte Idee. Du müsstest es halt etwas anpassen für private Nutzer 🙂 Ich kann mir vorstellen das auch mehr kommerzielle Anfragen kommen. Du hast ja damals auch die "Essentials" als regulären Auftrag gemacht. Ich bleibe auf jeden Fall dabei 👍🏻

ledermann, 2025-04-16:
Feature ist implementiert und kann im Beta-Test ausprobiert werden:
https://github.com/orgs/solectrus/discussions/4152#discussioncomment-12851682

PR #4195 (2025-05-19): Multiple inverters
Closes #3854 
Closes #1756