solectrus/solectrus #1769: Calculating bug

bennysweb, 2023-02-27:
Hi Georg, maybe I've found a bug in the expectation calculation or I don't understand the calculation correctly, also possible ;-)
In January, my system generates 27% less energy than expacted. In February till today 6% more. In my opinion this means my system has generated 21% (27% - 6%) less energy than expacted this year. Solectrus means my system has generated 6% less energy than expacted this year so far. Which value is now correct?

[image]
[image]
[image]

ledermann, 2023-02-28:
Hi Benny, thank you for the feedback and attentive observation. However, I do not believe it is an error, as the calculation basis varies each month.

Let me explain. I used the values from your screenshots, calculating the expected values backwards:

- January: expected `286`, actual `209` (`-27 %`)
- February: expected `400`, actual `424` (`+6 %`)

=> Year: expected `286 + 400 = 686`, actual `209 + 424 = 633`
=> `633 / 686 = 0,922`
=>`8 %` less than expected

Your annual screenshot shows `6 %` less than expected. Okay, there is still a (small) difference due to rounding errors or caching, which I will comment in #1770.

But fundamentally, I believe SOLECTRUS calculates correctly. One cannot simply add `-27 %` and `+6 %` to get `-21 %` because the reference value (i.e., expected value per month) is different each time.

bennysweb, 2023-02-28:
Thank you very much for sorting this out. Now I understand the calculation.

closed 2023-02-28 by bennysweb