solectrus/solectrus #1815: Temp. by 60Grade

StMulder, 2023-03-13:
The temp in Version 9.1 stands always at 60Degree.

Take a ook at www. Solardash.de

StMulder, 2023-03-13:
Case Temp laut log.html liegt tatsächlich bei 60Grad. Gehäuse ist aber recht kalt. Komisch

StMulder, 2023-03-13:
Der Fehler liegt garantiert nicht bei Solectrus.
Die anderen Werte des sind absolut realistisch 34/41 Grad. Ich schaue mal ob ich da eine Fühler finde.

Habe aktuell auch noch Tibber und lade den Speicher dann in Preisgünstigen Phasen. Heute z.B. hatten wir 14ct/kWh

ledermann, 2023-03-13:
Ok, dann verfolge ich das mit der Temperatur nicht weiter.

Ja, Tibber ist eine interessante Idee - die allerdings nicht so recht mit der Kostenkalkulation von SOLECTRUS kompatibel ist. Die aktuellen Preise müssten per API automatisch reinkommen. Mal sehen, was die Zukunft bringt, an Ideen mangelt es nicht  :)

closed 2023-03-13 by ledermann