solectrus/solectrus #1820: Werte anzeigen in der Grafik

Zeppelin-PV, 2023-03-14:
hallo Georg
möchte man die Werte in den Grafiken anzeigen, muss man ziemlich exakt der Konturlinie in der Grafik folgen.
Ist es möglich dass nach einmaligem "einfangen" die Werte auch dann angezeigt werden wenn man die Maus innerhalb der Grafik
nur noch horizontal bewegt, und die Werte quasi der Kontur folgen ?
LG Ernst

ledermann, 2023-03-14:
Berechtigter Hinweis, danke!

Das verwendete Tool für die Anzeige von Diagrammen ermöglicht das auch - es musste nur aktiviert werden, was ich nun gemacht habe. Auf https://solectrus.ledermann.dev ist es bereits zu sehen.

Die Änderung wird im nächsten regulären Update enthalten sein.

Zeppelin-PV, 2023-03-19:
... das sieht bei Dir schon mal ganz gut aus, aber:
fährt man die Grafik langsam ab, bleiben in der Konturlinie einzelne Messpunkte stehen (Punkt)
die Messwertanzeige (Box) sitzt zu nach am Messpunkt, Der Zeigerpfeil ist ungünstig kurz.
Die Anzeige sollte etwas weiter weg platziert werden.
Dann macht auch das Springen von rechts nach links in der Grafikmitte nichts aus.

ledermann, 2023-03-20:
Danke für die erneuten Hinweise! 

Das mit den "stehenbleibenden" Messpunkten konnte ich zwar nicht reproduzieren, aber den Pfeil habe ich nun vergrößert und die ganze Box etwas weiter weg platziert. 
Auf meiner Installation ist die Änderung zu sehen.

ledermann, 2023-03-30:
von @Zeppelin-PV in #1054:

> Die Anzeigebox ist ja nun mittlerweile etwas vom Messwertpunkt weggerückt. Gut so.
> Der Messwertpunkt ist in der gleichen Farbe wie die Grafik. Kann man den Messpunkten nicht zumindest einen schwarzen Rand verpassen, oder womöglich in einer Konträrfarbe darstellen ?

Ja, auch das ist einfach möglich. Einen schwarzen Rand um den Messpunkt habe ich gerade ergänzt.

ledermann, 2023-04-01:
Die soeben veröffentlichte Version v0.10.0 enthält die Änderung.