solectrus/solectrus #1822: Werte in kurzen Graphbalken Top 10

Zeppelin-PV, 2023-03-14:
hallo Georg
in den Top 10 Listen sowohl aufsteigend als auch absteigend werden bei kurzen Graphbalken keine Werte angezeigt.
Ist es möglich die Werte rechts des Balkens anzuzeigen ?
Die Skalierung der Balken müsste ggf. geändert werden, so dass bei allen immer rechts davon Platz ist.
... einfach so ne Idee
LG Ernst

ledermann, 2023-03-15:
Ja, korrekt, das kann verbessert werden. Bei kleinen Balken Zahl rechts daneben, bei großen Werten Zahl im Balken (wie jetzt). Mache ich bei Gelegenheit.

ledermann, 2023-03-21:
Die Werte in der Top-10 werden jetzt je nach Größe des Balkens entweder innerhalb oder außerhalb angezeigt. 
Die Änderung wird im nächsten regulären Update enthalten sein.

ledermann, 2023-04-01:
Die soeben veröffentlichte Version v0.10.0 enthält die Änderung.