solectrus/solectrus #1851: Temperatur Push Meldung (Brandgefahr)

Zeppelin-PV, 2023-03-26:
... hallo Georg
alle Senec Besitzer haben, auf Grund einzelner Vorfälle, etwas Bedenken in Bezug auf Brandgefahr der Lithium Ionen Zellen.
Ein thermal runaway ist keinem zu wünschen. Solectrus hat bereits eine Temperaturanzeige. Das Dashbord des Speichers zeigt zwei
Temperaturen an. (Lade- bzw. Wechselrichter)
Als momentan Betroffener (Anlage ist auf 50% gedrosselt) mache ich mir natürlich auch Sorgen.
Meine Frage bzw. Vorschlag:
ist es möglich die sensiblen Temperaturen zu überwachen und im Falle eines nicht natürlichen Wertes eine Push Nachricht (smtp) an eine frei wählbare eMail Adresse abzusetzen ?
Das wäre ein ganz großer Schritt in Sachen vorbeugende Sicherheit.

LG Ernst

ledermann, 2023-03-26:
Ja, so etwas in der Art wäre schon nützlich. SOLECTRUS übermittelt derzeitig die Gehäuse-Temperatur (und auch nur diese, also keine Akku-Temperaturen oder was sonst noch als Indikator taugen würde). Dieser Wert wird in die InfluxDB geschrieben und von SOLECTRUS angezeigt. 

Automatische Push-Benachrichtigungen sind allerdings technisch aufwändiger. Derzeitig gibt es in SOLECTRUS auch gar keinen Hintergrundprozess, der etwas überwachen könnte. SOLECTRUS wird derzeitig nur dann aktiv, wenn man es im Browser geöffnet hat. Aber das ist natürlich machbar, langfristig zumindest.

Kurzfristig gibt es aber auch jetzt eine Möglichkeit, derartige Benachrichtigungen zu erhalten. InfluxDB bringt das nämlich bereits mit, auch wenn es recht kompliziert einzurichten ist und auch nicht mit E-Mail funktioniert, sondern nur Slack und HTTP-Endpoints. 

Grob geht das wie folgt:

1. Sicherstellen, dass InfluxDB von außen erreichbar ist. Dazu muss in der `docker-compose.yml` der Port `8086` aufgeführt werden, siehe hier: https://github.com/solectrus/hosting/blob/main/guide/raspberry-pi/docker-compose.yml#L48-L50
2. Einloggen bei InfluxDB unter `raspi-ip:8086` mit `$INFLUX_USERNAME` und `$INFLUX_PASSWORD` (siehe `.env`-Datei)
3. Alert anlegen, bestehend aus Check, Notification endpoint und Notification rule (weitere Infos hier: https://docs.influxdata.com/influxdb/v2.6/monitor-alert/)

Ich habe mir damit genau so einen Temperatur-Warner gebastelt, der mich bei Überschreiten eines Temperatur-Grenzwertes auf Slack benachrichtigt. Das ist aber äußerst fummelig gewesen. So etwas direkt in SOLECTRUS zu haben, wäre schon gut.

TheReal1988, 2023-04-08:
Hallo Ledermann,

zunächst einem vielen Dank für das tolle Projekt, es ist wirklich hilfreich!

Ich habe Probleme, über den Browser und über ioBroker auf die InfluxDB zuzugreifen. Gemäß deines Kommentares habe ich in der docker-compose.yml den Port 8086 aufgeführt.
`  # Optional: Allow InfluxDB to be accessed from the outside.
     ports:
      - 8086:8086
`
Ich kann aber aber leider nicht über die Raspi IP Port 8086 (http://192.168.1.25:8086/) auf die InfluxDB zugreifen, ich bekomme nur "Die Webseite ist nicht erreichbar" angezeigt.
Woran könnte das noch liegen?

Und im Solectrus Dashboard bekomme ich unter "Jetzt" und "Top 10" alle Werte angezeigt, wenn ich auf Tag, Woche, Monat oder Jahr wechsle, wird mir "Content missing" angezeigt. Ich habe alle PV Daten der vergangenen 12 Monate via CSV importiert, diese werden unter den Top10 auch korrekt angezeigt.

Vielen Dank vorab und LG
Dominik