solectrus/solectrus #1859: Feature Request: Eigenes Farbschema

noffycws, 2023-03-28:
SOLECTRUS kommt ja mit voreingestellten Farben daher. Ich fände es schön, wenn man die Darstellung selbst ändern könnte.

Vielleicht zunächst die Möglichkeit für einen Darkmode, den man entweder selbst auswählt oder die Tageszeit abhängig ein- und ausgeschaltet wird.

Ebenso verbunden mit der Möglichkeit die Farben in den Anzeigen und Grafen selbst zu wählen. Bspw. "gelb" für die Erzeugung o.ä.

Nur als Idee und Feature-Request.

ledermann, 2023-03-28:
Ja, einen Darkmode finde ich auch sinnvoll.

Von frei definierbaren Farben halte ich aber nicht so viel, weil das ein erheblicher Implementierungsaufwand wäre und der User damit auch sehr hässliche Dinge erzeugen könnte ;-)

Ggfs. kann ich die verwendeten Farben aber überdenken. Gelb für die Erzeugung ist zwar naheliegend, allerdings wirkt dann alles gleich recht bunt, was ich eher vermeiden möchte.

noffycws, 2023-03-30:
Naja, hässlich ist ja immer im Auge des Betrachters ;-) Ich finde gelb für die Erzeugung recht schön ;-)

Vielleicht kann man ja ein Farbschema definieren (4 - 5 Farben) und die dann umsetzen. So muss/kann man nicht jeden Farbwert durch den Nutzer setzen lassen können. Aber man hat die Möglichkeit die eigene SOLECTRUS-Instanz etwas zu individualisieren.

Zeppelin-PV, 2023-04-02:
hallo Georg, großer Wurf die 10.0, super !
was vielleicht auch zu diesem Thema passt ist folgendes: (nicht schon wieder ein neues issue)
die Grafikfelder erscheinen, je nach Einfärbung, globik, schwer.
Wäre es machbar hier von den Werten (dunkle Linie) hin zur Null-Linie ein Ausbleichen der Farbe zu gestalten ?
Somit käme die Grafik viel lockerer daher. 
Bei den Top Ten Balken ist dies ein wenig ersichtlich.
Ist ja gewissermaßen auch ein Punkt zum Farbschema.
LG Ernst

StevieC121176, 2023-06-10:
Nachdem ich jetzt einige Wochen Solectrus nutze finde ich das Farbschema ganz ansprechend. Was allerdings aufkommt ist der Wunsch nach einem Darkmode, bei dunklen Lichtverhältnissen ist das weiß schon arg präsent sag ich mal. Vielleicht würde sich der Aufwand für einen Darkmode ja in Grenzen halten.

ledermann, 2024-09-02:
Work in Progress :)

[image]

StevieC121176, 2024-09-03:
Wow , schaut sehr gut aus. Genau das richtige für unser Kühlschrank Display.

StevieC121176, 2024-09-05:
Wirklich sehr gut gelungen der Dark Mode. Eine Kleinigkeit evtl. noch. Unter iOS im Safari Browser wird der obere Teil immer sehr hell dargestellt, im dark und Normalmode in der gleichen Farbe. Könnte man das im Darkmode noch anpassen?

[image]

ledermann, 2024-09-05:
Der @StevieC121176 entdeckt die neuen Sachen immer schneller, als dass ich sie beschreiben kann ;-)

Ja, der Dark-Mode steht jetzt im `next` Branch in einer noch **sehr groben** Fassung bereit. Wer möchte, kann das ausprobieren. Einige Details sind es aber noch nicht stimmig, da werde ich weiter dran feilen. Die Zeile ganz oben gehört dazu.

StevieC121176, 2024-09-06:
Vielen Dank, jetzt paßt das mit der oberen Zeile sehr gut :-)

gereons, 2024-09-06:
Mir gefällt's im großen und ganzen schon ziemlich gut 👍 

Details die mir aufgefallen sind:

Der "Transport"-Balken in "Aktuell" sieht im iOS-Safari noch komisch aus.
Ein nicht verbundenes E-Auto würde ich eher durch ein dunkleres Icon symbolisieren als durch ein helleres (ggü dem Haus-Icon)
Der fette fast schwarze Rahmen um die Balken links zieht sehr viel Aufmerksamkeit da hin. Evtl. die Hintergrundfarbe und die Balkenfarbe tauschen?

StevieC121176, 2024-09-06:
Mir ist noch eine Kleinigkeit aufgefallen. Im Tooltip wird der Betrag in grün angezeigt. Sollte analog der anderen Kosten eher rot sein?

[image]

ledermann, 2024-09-06:
Danke für die Rückmeldungen, ich habe das eingearbeitet und auch noch andere Details verbessert und korrigiert. 

Erwähnenswert: Die Auswahl habe ich ins Menü verschoben (auch dort: noch nicht perfekt). Es wird aber jetzt zwischen Hell/Dunkel/Automatisch unterschieden. Default ist "Automatisch", übernimmt also die Systemeinstellung. Wenn man grundsätzlich oder tageszeitabhängig auf seinem Gerät im Darkmode unterwegs ist, muss man in SOLECTRUS also gar nichts einstellen.

Ansonsten gefällt mir selbst noch nicht alles. Manches ist mir noch zu kontrastarm oder erscheint mir nicht ideal, auch wenn ich nicht weiß, was genau mich stört. Erst mal sacken lassen :-)

StevieC121176, 2024-09-07:
E-Auto hat jetzt die gleiche Farbe wie der Rahmen.

[image]

gereons, 2024-09-07:
Auf allen meinen iOS-Geräten sieht der Verbindungsbalken bei "Aktuell" noch komisch aus:

[image]

Im macOS-Safari ist alles in Ordnung, und da es bei @StevieC121176 vernünftig aussieht scheint es mit irgendwelchen lokalen Settings zusammenzuhängen, aber ich hab keine Ahnung welche das sein könnten...

gereons, 2024-09-07:
Was mir noch gerade auffällt: die Hairlines in den Charts sind unsichtbar geworden

ledermann, 2024-09-07:
@gereons Bei mir auf iOS 17.6.1 sieht die Flow-Animation auch korrekt aus. Vielleicht mal die Web-App explizit neu starten?

gereons, 2024-09-07:
Ah, hab's gefunden. "Bewegung reduzieren" in den Accessibility-Einstellungen ist die Ursache - wenn man das auf dem Mac anmacht sieht's da auch genauso aus.

gereons, 2024-09-08:
Wenn in den Top 10 die Zahl nicht mehr im Balken sondern rechts daneben steht, kann man sie kaum lesen:

[image]

Ansonsten ist die einzige Sache die für meinen Geschmack noch angepasst werden könnte die "Erzeugung" in der Tagesansicht - da ist IMHO die Prognose viel zu hell, die sollte ggü der Erzeugung mehr in den Hintergrund rücken.

ledermann, 2024-09-09:
Prima, danke für das erneute genaue Hinschauen. Alle oben erwähnten Punkte sind jetzt korrigiert.

gereons, 2024-09-09:
Super, danke!

ledermann, 2024-09-11:
Mittlerweile habe ich weitere Details korrigiert und dem Theme-Switcher ein besseres Aussehen verpasst. Ich denke, dass ich nun durch bin mit diesem Thema.

Ein Problem gibt es aber, vorerst ohne Lösung: Wenn man SOLECTRUS unter iOS auf den Home-Screen legt und es beim Default belässt (d.h. "Automatisches Farbschema"), dann reagiert die App fälschlicherweise **nicht** auf systembedingte Wechsel zwischen Hell/Dunkel. Dies scheint ein [Bug in iOS](https://forums.developer.apple.com/forums/thread/739154) zu sein. Es soll einen abenteuerlichen Workaround geben, den ich aber nicht implementiert bekomme.

Da der Bug möglicherweise mit iOS 18 behoben sein wird, warte ich einfach mal auf [nächsten Montag](https://www.heise.de/news/iOS-18-macOS-15-Co-Wann-Apple-die-neuen-Betriebssysteme-veroeffentlicht-9862783.html) :-)

gereons, 2024-09-11:
Mir gefällt's richtig gut, danke für die Implementation.

Ein winziges Detail ist mir neulich noch und auch nur im Dark Mode aufgefallen: die Farbgebung der Mouseover ist zwischen Chart und Balken invertiert, sprich einmal dunkel-auf-hell und einmal hell-auf-dunkel. Sähe das nicht (noch) besser aus wenn es konsistent wäre und für beide die gleichen Farben verwendet würden?

ledermann, 2024-09-11:
@gereons Berechtigter Hinweis, danke! Ich habe jetzt die Diagramm-Tooltips (Chart.js) an die sonstigen Tooltips (Tippy.js) visuell angeglichen. Das ergibt zwar keine 100%ige Übereinstimmung, da es sich um verschiedene Libraries handelt, aber die Tooltips sind nun alle gleichermaßen hell bzw. dunkel.

@StevieC121176 Interessant. Ich warte mal ab, wie sich das finale iOS 18.0 verhalten wird und prüfe dann, ob ich noch etwas verbessern kann.

ledermann, 2024-09-17:
iOS 18 ist da und wie es mir scheint, ist darin tatsächlich der Fehler behoben: Eine PWA (wie SOLECTRUS) wird über Änderungen des System-Erscheinungsbildes (Hell/Dunkel) informiert und kann entsprechend reagieren. Somit ist jetzt alles fein :-)

closed 2024-09-21 by ledermann

ledermann, 2024-09-21:
Ich denke, der Darkmode ist nun ausreichend getestet, die Änderungen aus PR #3432 habe ich daher nach `develop` übernommen.

ledermann, 2024-09-23:
Released als Teil von v0.17.0

PR #3432 (2024-09-21): Dark mode
Resolves #1859