solectrus/solectrus #1881: Diagramme mit Farbverlauf

ledermann, 2023-04-02:
von @Zeppelin-PV in #1859

> die Grafikfelder erscheinen, je nach Einfärbung, globik, schwer.
> Wäre es machbar hier von den Werten (dunkle Linie) hin zur Null-Linie ein Ausbleichen der Farbe zu gestalten ?
> Somit käme die Grafik viel lockerer daher.
> Bei den Top Ten Balken ist dies ein wenig ersichtlich.

StevieC121176, 2023-06-19:
Vielleicht könnte man es so in der Art machen, Bild 1 ist mit Farbverlauf hell und dunkle Umrandung , Bild 2 helle Farben und dunkle Umrandung.

[image]

[image]

ledermann, 2023-06-20:
Ja, das sieht gut aus. Ich verwende für die Diagramme [Chart.js](https://www.chartjs.org/) und bin an die davon angebotenen Möglichkeiten gebunden. Aber ein Farbverlauf erscheint machbar: http://jsfiddle.net/4vobe59a/

StevieC121176, 2023-06-20:
Das Beispiel aus dem Link schaut gut aus.Farbverlauf und die stärker hervorgehobene Kante ist dort auch zu sehen

ledermann, 2023-06-22:
Das habe ich jetzt mal umgesetzt. Sämtliche Diagramme erscheinen nun mit einem Farbverlauf.

Dies wird im nächsten Update enthalten sein, in der [Live-Demo](https://solectrus.ledermann.dev) ist es ab sofort zu sehen.

closed 2023-06-22 by ledermann

StevieC121176, 2023-06-22:
Vielen Dank, gefällt mir sehr gut.

StevieC121176, 2023-07-28:
Eine Kleinigkeit zur Optik: Bei der Anzeige Verwendung und Herkunft sind die Ecken leicht abgerundet dargestellt. Vielleicht wäre das bei den Balkendiagrammen auch möglich und würde diese etwas "weicher" aussehen lassen. Als Beispiel hier die Small rounded Balken bei Charts.js. https://www.chartjs.org/docs/latest/samples/bar/border-radius.html