solectrus/solectrus #1886: GUI - Widersprüchliche Angaben

GuenniAC, 2023-04-03:
Unterschiedliche Angaben für Hausverbrauch wenn "Mouse-over" Feld, siehe Anhang
[image]

ledermann, 2023-04-03:
Ja, das ist ein Darstellungsfehler. Danke fürs Melden!

Ursache ist die automatische Aktualisierung per AJAX alle 5 Sekunden im Browser, die für Tooltips nicht so recht funktioniert, d.h. der vorherige Tooltip bleibt bestehen. Mache einen Full-Reload (Browser-Refresh), dann stimmt es wieder.

Ich werde das korrigieren.

ledermann, 2024-03-09:
Dieser Bug ist nun endlich behoben, die Tooltips werden jetzt bei jeder neuen Messung mit aktualisiert.

Die Korrektur fließt ins kommende Update mit ein. In der [Live-Demo](https://demo.solectrus.de/house_power/now) ist sie bereits enthalten.

closed 2024-03-09 by ledermann