solectrus/solectrus #1887: Better scale for temperature chart

bennysweb, 2023-04-03:
Congrats on version 0.10, good job! The next coffee is on me ;-)

Can you improve the visualization of the daily/monthly/yearly temperature chart with an adaptive y axis like the generation charts? The temperatures are mostly in a range of 20 - 40 degrees, half of the displayed scale is actually not used.

ledermann, 2023-04-03:
Thanks :) 

Yes, better scaling is a good idea, I will do this for the next update.

ledermann, 2023-04-26:
@Zeppelin-PV Danke für den Input! 

Ich versuche, eine möglichst minimalistische Darstellung zu erreichen und lasse gerne unwesentliche Informationen weg. Dazu gehört z.B. die Nachkommastelle bei der Temperatur, dargestellt wird stattdessen ein gerundeter Wert. Auch glaube ich nicht, dass der Temperatur-Sensor so präzise ist, dass er auf Zehntel-Grade genau messen kann. Ich möchte keine Genauigkeit suggerieren, die nicht gegeben ist - selbst wenn SENEC das tut :)

Dennoch: Das Diagramm habe ich kürzlich verbessert, sodass dort der Originalwert (also nicht gerundet) verwendet wird. Das sorgt tatsächlich für eine fließendere, weniger zackige Darstellung:
https://solectrus.ledermann.dev/case_temp/2023-04-25

Verfügbar mit dem nächsten Update.

Zeppelin-PV, 2023-04-26:
hallo Georg,
die Anzeige im "Jetzt" in ganzen Grad ist ok, 
in Deinem Link werden ja die Zehntel-Grad angezeigt wenn man die Grafik abfährt
bei "Tag" sogar dreistellig (?), würde eine Kommastelle auch reichen
... so ist es gut

Zeppelin-PV, 2023-04-26:
in den Balkengrafiken Woche, Monat etc. werden auch dreistellige Kommastellen angezeigt

ledermann, 2023-05-04:
Drei Nachkommastellen sind natürlich nicht nötig. Im Diagramm erfolgt die Rundung der Temperatur jetzt immer auf **eine** Nachkommastelle.

closed 2023-05-04 by ledermann

ledermann, 2023-05-06:
Im Rahmen von Version v0.10.1 ab sofort verfügbar

hambue, 2023-05-06:
Meine Tiefsttemperaturen lagen im März und April bei 18.6 °C. In der Jahresübersicht sind diese Temperaturen nun abgeschnitten durch die 20 °C Grenze. 

[image]

Dies stellt selbstverständlich kein großes Problem dar, aber vielleicht wäre es möglich statt der starren unteren Grenze, den Minimalwert aus der abgerundeten Tiefsttemperatur abzuleiten. 

PS: Vielen Dank für die tolle App. Sie läuft seit Inbetriebnahme im März problemlos auf einem zu einem Ubuntu-Server umgerüsteten alten Laptop.