solectrus/solectrus #1891: Kosten: Real vs. kalkulatorisch

ledermann, 2023-04-04:
Die derzeitige Darstellung der kalkulatorischen Kosten ist möglicherweise etwas irritierend:

[image]

Alternativ oder ergänzend sollte die realen Kosten, also nur die durch den Strombezug entstandenen Kosten, dargestellt werden.

ledermann, 2023-04-15:
Vorschlag eines Users:

> Gib an, wie hoch die Stromkosten sind, die man real zu bezahlen hat. Also ausschließlich den Netzbezug. Diese Kosten sind greifbar und fallen ja wirklich an. Du könntest beispielsweise schreiben:
>
> Gesparte Stromkosten (durch Eigenverbrauch): 194€
> Erhaltene Einspeisevergütung:   66€
> -> Einsparung:   260€
> Anteil Akku: 44€
> Kosten durch Netzbezug: 36€
>
> Solectrus gibt für März mit den gleiche Werten das hier aus (Bild)
[image]

ledermann, 2023-05-26:
Zwischenzeitlich habe ich die Kostenberechnung gegenüber der aktuellen Version v0.10.1 grundlegend überarbeitet. Diese sieht nun so aus, hier am Beispiel eines ganzen Jahres:

[image]

Zu sehen ist das auf meiner Installation, wo der `develop`-Branch läuft:
https://solectrus.ledermann.dev/grid_power/2022

Die Änderungen am Popup des Badges "Kosten" im Einzelnen:
- Es werden keine **kalkulatorischen** Kosten mehr dargestellt, sondern nur noch die **real** entstandenen Kosten. Das sind diejenigen, die aufgrund des Netzbezuges tatsächlich an den Stromversorger bezahlt werden müssen. Eine entgangene Einspeisevergütung ist also nicht mehr im Spiel.
- Wenn es im gewählten Zeitraum mehrere Strompreise gab, werden diese als Intervall (kleinster Preis - größter Preis) dargestellt. Damit kann man das zwar immer noch nicht genau nachrechnen, intern wird aber tagesgenau gerechnet. Eine präzise Aufschlüsselung der Berechnung erfolgt derzeit nicht. Grundsätzlich habe ich die Tendenz, alles nicht unbedingt Erforderliche eher wegzulassen, um das UI einfach zu halten.

Es gibt weiterhin den Badge "Einsparung". Der ist unverändert, erwähnt aber ebenfalls die Kosten. Und das sind nun die gleichen wie im Badge "Kosten", was vorher (also noch in v0.10.1) anders und somit irritierend war.

@GrimmiMeloni Grundsätzlich liegt mir die richtige Berechnung sehr am Herzen. Ich habe da bereits viel Feedback per E-Mail bekommen und mit einigen Usern größere Excel-Tabellen ausgetauscht und leidenschaftliche Diskussionen geführt ;-) 
Ich hoffe, dass die jetzige Lösung, die ich seit drei Wochen auf meiner Installation ausprobiere, in die richtige Richtung geht.

Sehr gerne können wir etwaige Nachbesserungen oder grundsätzliche Veränderungen diskutieren.

GrimmiMeloni, 2023-05-26:
Ich hab mir den develop build mal deployed. 

Für den Kosten Badge, macht der Change so jetzt Sinn.

Beim Einsparung Badge wäre es jetzt wichtig zu verstehen wie sich die Werte "Einsparung" und "Vergleichspreis" genau berechnen. Wie oben geschrieben -  die "Kosten" des Eigenverbrauchs sollten in die Berechnung des Werts für die Einsparung einfließen. Es ist ja Fakt, das durch Eigenverbrauch Opportunitätskosten entstehen. Wenn ich 1kWh Eigenverbrauch habe, dann spare ich einerseits den Bezugspreis, im Gegenzug entgehen mir aber auch 8ct Einspeisevergütung. Daher ist die Ersparnis durch Eigenverbrauch meiner Ansicht nach Bezugspreis abzgl. Eispeisevergütung pro kWh. Wenn es so bereits berechnet wird, passt es, sonst wird (meiner Ansicht nach) die Ersparnis zu hoch berechnet.

Weiteres Feedback zum Tooltip - ich finde positiv/negativ Darstellung irritierend.

[image]

Mir ist nicht ganz klar was jetzt genau berechnet wird. Die Zahlen suggerieren mir das hier Solarpreis und Vergleichspreis aufaddiert werden. Da Vergleichspreis negativ ist, passt das aber irgendwie auch nicht so recht. So ganz rund fühlt es sich noch nicht an...

ledermann, 2023-05-26:
Ok, danke für das Feedback! Vor einer möglichen Verbesserung möchte ich erst mal erklären, wie meine Berechnung verläuft und was der Grundgedanke ist. Das Thema kommt nämlich immer wieder zur Sprache.

Grundsätzlich: Ich vergleiche die aktuelle Situation (d.h. **MIT** Photovoltaik) mit einem theoretischen Szenario, bei dem man **KEINE** PV-Anlage hat, aber den exakt gleichen Stromverbrauch. Ich mache also einen Vorher-Nachher-Vergleich und bilde die Differenz, die ich als "Ersparnis durch die PV-Anlage" bezeichne.

Jetzt konkret zu Deinem im Screenshot ersichtlichen Beispiel. Es gibt drei Schritte:

1. MIT Photovoltaik: Es wird wenig Strom gekauft (hier €0,40) und es gibt eine Einspeisevergütung für den Überschuss (hier €20). Zusammen ergibt das den "Solarpreis" (doofes Wort) als Summe, wobei die Vorzeichen zu beachten sind (hier: -€0,40 + €20 = €19). Ja, Rundungsfehler sind nicht gut.

2. OHNE Photovoltaik: Der Strom für den gesamten Verbrauch (im Screenshot leider nicht sichtbar) muss gekauft werden und es gibt natürlich keine Einspeisevergütung. Gerechnet wird (Hausverbrauch + Wallbox) * Strompreis. In Deinem Beispiel ergeben sich Strombezugskosten von €35.

3. Die Ersparnis, die die PV-Anlage bringt, ist somit die Differenz der beiden Beträge. Weil der Solarpreis sowohl positiv (Sommer) als auch negativ (Winter) sein kann, sind die Vorzeichen wichtig. In Deinem Beispiel wird gerechnet:
€19 - (- €35) = €55. In Worten also: Dadurch, dass Du eine PV-Anlage hast, hast Du im gewählten Zeitraum €55 mehr im Geldbeutel, als wenn Du keine PV-Anlage hättest. Ja, die Investitionskosten für die Anlage habe ich weggelassen.

Man könnte sicherlich auch ganz anders rechnen und je nach Methode spielen dann auch die Opportunitätskosten eine Rolle, die grundsätzlich natürlich existieren. Bei dem von mir gewählten Vorher-Nachher-Vergleich spielen sie aber keine Rolle, denke ich.

Was ich dennoch hier verbessern sollte:

- [ ] Die drei Schritte visuell besser herausarbeiten
- [ ] Berechnung des Vergleichspreises darstellen
- [ ] Begriff "Solarpreis" durch etwas Besseres ersetzen
- [ ] Rundungsfehler dadurch beheben, dass alle Beträge mit Cents dargestellt werden. Erst ganz am Schluss kann die Ersparnis (bei größeren Beträgen) weiterhin gerundet werden.
- [ ] Eine negative Zahl sollte nicht gleichzeitig noch rot gefärbt werden. Es genügt, positive Zahlen grün darzustellen und negative rot. 

Aber bevor ich diese Verbesserungen angehe, sollten wir Konsens über die Berechnungsmethode herstellen 😄

GrimmiMeloni, 2023-05-26:
Danke erstmal für die ausführliche Erläuterung. Szenario soweit verstanden. 

> OHNE Photovoltaik: Der Strom für den gesamten Verbrauch (im Screenshot leider nicht sichtbar) muss gekauft werden und es gibt natürlich keine Einspeisevergütung. Gerechnet wird (Hausverbrauch + Wallbox) * Strompreis. In Deinem Beispiel ergeben sich Strombezugskosten von €35.

Erstmal - es wird richtig gerechnet. ;)
Und wenn das Vergleichsszenario eine Welt ohne PV (anstatt ohne Eigenverbrauch) ist, dann macht es auch Sinn den vollen Netzbezugspreis anzunehmen.

Zu den "Restarbeiten" noch ein paar Anmerkungen:

> * [ ]  Die drei Schritte visuell besser herausarbeiten
> * [ ]  Berechnung des Vergleichspreises darstellen

Vielleicht eher in einen FAQ Bereich verschieben, und nur dorthin verlinken? Ich empfinde die Optik eigentlich ganz aufgeräumt wie sie ist. Es reicht meiner Ansicht nach, wenn "irgendwo" nachvollziehbar ist, wo gerechnet wird. Theoretisch würde ja schon ein Link auf diese Diskussion hier ausreichen, aber könnte man auch etwas hübscher aufbereitet irgendwo parken.

> * [ ]  Begriff "Solarpreis" durch etwas Besseres ersetzen
Wie wäre es mit "Netzstrompreis", oder "Bilanz Netzbezug"?

> * [ ]  Rundungsfehler dadurch beheben, dass alle Beträge mit Cents dargestellt werden. Erst ganz am Schluss kann die Ersparnis (bei größeren Beträgen) weiterhin gerundet werden.

Nö. Auch hier - unnötig "noisy". So lange intern Cent-genau gerechnet wird, und sich Rundungsfehler nicht aufaddieren, ist es ok.

> * [ ]  Eine negative Zahl sollte nicht gleichzeitig noch rot gefärbt werden. Es genügt, positive Zahlen grün darzustellen und negative rot.

Nein. Bitte so lassen. Das ist in der Tat so gut und hilfreich. Ich glaube die Berechnungsformel zu dokumentieren sie 3 Punkte weiter oben ist das einzig wichtige. Wenn die Idee ist die Ersparnisse zu zeigen, dann ist es auch hilfreich, wenn Geld aus dem Haushalt abfließt, dies auch negativ UND rot darzustellen.

GrimmiMeloni, 2023-05-27:
Eine Zusatzfrage noch: Woraus wird der "Anteil Akku" in den Einsparungen berechnet? Hatte `Akkuentnahme / Verbrauchswerte` erwartet, daß kommt aber nicht hin.

ledermann, 2023-05-27:
Sehr gut! Freut mich, dass wir uns bei der Berechnung einig sind.

Den Hinweis mit der FAQ finde ich sehr sinnvoll, das werde ich machen und dann mit einem kleinen Icon darauf verweisen (jeweils im Badge "Kosten" und "Ersparnis").

Beim Ersatz für den Begriff "Solarpreis" zögere ich noch. Noch finde ich nicht, dass die Vorschläge unmissverständlich klarmachen, dass es sich um den jetzigen effektiven Betrag handelt, der aufgrund der PV-Anlage entsteht und sowohl positiv als auch negativ sein kann. Naja, entscheidend ist, dass es in der FAQ vernünftig erklärt wird.

Zum "Anteil Akku": Der wird wie folgt berechnet:

```
   Ersparnis durch Akku (€) = 
       Akkuentnahme (kWh) * Strompreis (€/kWh) 
     - Akkubeladung (kWh) * Einspeisevergütung (€/kWh)
```

Die Idee ist also, dass man den aus dem Akku entnommenen Strom nicht kaufen muss, also den Strompreis einspart. Allerdings verliert man auch die Einspeisevergütung für die Strommenge, mit der man den Akku auflädt. Die Akku-Ersparnis ist also der Betrag, den ich spare, weil ich einen Akku habe - gegenüber der Situation, eine PV-Anlage ohne Akku zu betreiben. Mit diesem Betrag amortisiert sich also die Investition in den Akku.
Den Betrag setze ich dann prozentual ins Verhältnis zur vorher ausgerechneten Gesamtersparnis.

ledermann, 2023-06-03:
Hier habe ich kaum noch noch etwas verändert, weil ich auch denke, dass es so passt. Für die Erklärung der Berechnungen habe ich aber zwei FAQ-Artikel hier eingestellt:

https://solectrus.de/faq/kosten-berechnung/
https://solectrus.de/faq/ersparnis-berechnung/

Ich hatte erst ein Info-Icon mit Links auf diese Artikel in die beiden Badges integriert, das war aber nicht ideal, weil immer sichtbar. Im Tooltip wäre es besser aufgehoben, aber das kollidierte mit dem automatischen Schließen des Tooltips bei Mausbewegung. Als Ersatz gibt es nun einen Link zur FAQ von der Hauptnavigation aus. Das sollte genügen.

Damit schließe ich dieses Issue.

closed 2023-06-03 by ledermann