solectrus/solectrus #1894: Top10-Wert für aktuellen Zeitraum manchmal ungenau

ledermann, 2023-04-05:
### Discussed in https://github.com/orgs/solectrus/discussions/1888

Wenn der aktuelle und noch nicht beendete Tag, Monat, Jahr in den Top-10 vorkommt, ist dessen Wert manchmal etwas zu hoch. Ursache ist vermutlich die verwendete InfluxDB-Abfrage, die zu einer Hochrechnung führt.

ledermann, 2024-07-03:
Ursache ist, dass mit Flux die Verwendung von `aggregateWindow(every: 1h, fn: mean)` ungenau ist, wenn der Zeitraum noch nicht abgeschlossen ist.

Lösungsansatz: 

- Statt einer Top10 intern eine Top11 berechnet
- Wenn auf den ersten 10 Plätzen der aktuelle Zeitraum vorkommt, diesen zusätzlich separat berechnen (über `PowerSum`), Wert austauschen, neu sortieren (alle 11) und die ersten 10 nehmen

AlpenFlizzer, 2024-08-28:
Ich stolpere selbst gerade wieder über diesen Umstand. 
Ich habe eine neue Solectrus Instanz die heute den ersten Tag speichert.
In allen Timeframe Tabs werden verschiedene Werte angezeigt. Ich verstehe, dass es an unvollständigen Tagen wegen der Flux zu ungenauem Verhalten kommt, aber bedeutet das, dass ich jetzt warten muss, bis diese Woche, Monat, Jahr, zukünftige Jahre immer erst vollständig sein müssen, damit sie korrekt dargestellt werden? Oder heißt das bei meiner neuen Instanz, dass ab morgen die Anzeigen korrekt sind, wenn der erste Tag im Kasten ist?

Eventuell könnte man den Umstand noch in FAQs aufnehmen/beschreiben, da der Weg zu dem Issue recht verschachtelt war und es mehrfach-Issues zum selbenm Thema, wenn auch mit Abwandlungen, gibt. Ich hätte auch keine Erwähnung entdeckt, wie weit eine Lösung entfernt ist oder wie nah.

Danke für eure Hilfe!

ledermann, 2024-08-28:
@AlpenFlizzer Meinst du wirklich die Top10? Wenn ja: Das ist leider ein bekanntes Problem, eine Lösung habe ich oben (3. Juli) skizziert, aber noch nicht implementiert.

Es ist so: Wenn deine Instanz heute den ersten Tag aufgezeichnet hat (aber noch nicht vollständig, da der 28. August noch nicht vorbei ist) und du die Top10 für Tag aufrufst, dann wird diese nicht exakt den richtigen Wert anzeigen. Rufst du dann morgen die Top10 für Tag auf, wird der 28. August korrekt angezeigt. Dann wird jedoch der Wert vom 29. August nicht exakt sein. So geht es weiter, bis irgendwann der aktuelle Tag nicht mehr in der Top10 vorhanden ist.

Am Anfang, wenn die Instanz erst seit kurzem läuft, hat die Top10 sowies keine besondere Aussagekraft. Mit der Zeit justiert sich das dann. Nach einer gewissen Zeit ist die aktuelle Periode sowie nur noch sehr selten enthalten, das Problem verschwindet dann quasi.

Dennoch ist eine Korrektur natürlich noch geplant. Ggfs. könnte ich auch eine einfachere Lösung implementieren und die aktuelle Periode grundsätzlich gar nicht in der Top10 anzeigen. Also so, wie es auch bei der Flop10 gemacht wird.

Korrekt ist übrigens immer die Strombilanz, also wenn du dort den betreffenden Zeitraum aufrufst. Das ist der Vergleichsmaßstab.

AlpenFlizzer, 2024-08-28:
Ich meine die Strombilanz, habe mich bloß wegen der Artverwandtschaft an diesen Issue angehängt.
Ich zeige mal das Problem:
[image]

Wie schon erwähnt ist das eine Solectrus Instanz die ich seit heute frisch füttere. Nach einigen Setup Problemen habe ich den Influx Container und die Daten im Bind-Mount gelöscht und mit der Speicherung neu begonnen. Kann das ein Probem sein?
Das Bild ist demonstrativ, ich habe ähnliche Abweichungen bei den anderen Sensoren auch.

Die Daten kommen alle 5 Sekunden via MQTT daher. Ich hatte bei meiner ersten Instanz ebenfalls komische Probleme mit den historischen Daten aus Solaredge, damals lag es daran, dass die Datenpunkte wohl nicht zu einem erwarteten Zeitintervall gepasst haben.

PR #3533 (2024-11-07): Auswertungen anhand vorberechneter Tageswerte
Dieser Pull-Request behebt verschiedene Probleme auf einmal:

- Keine Abweichungen mehr zwischen Diagrammen und Strombilanz (resolves #2032)
- Keine Abweichungen mehr zwischen Top10 und Strombilanz (resolves #1894)
- Keine Fehlberechnungen mehr bei Lücken in den Messwerten (wird immer wieder mal thematisiert, z.B. #2654)

Außerdem wird die Performance von Auswertungen (Strombilanz, Diagramme, Top10 und Kacheln) für längere Zeiträume deutlich verbessert. Die Abfragen sind um den Faktor 50 bis 100 schneller.

Wie diese Verbesserungen erreicht werden: [... cut]
closed 2024-11-07 by ledermann