solectrus/solectrus #1896: Messpunkte in Grafiken bleiben stehen

Zeppelin-PV, 2023-04-06:
Fährt man eine Grafik mit dem Cursor langsam ab bleiben vereinzelte Messpunkte stehen.
Sowohl mit Firefox als auch mit Chrome (Windows)

[image]

ledermann, 2023-04-07:
Ja, dieses kosmetische Problem tritt auf der JETZT-Seite auf, wenn ein neuer Messpunkt ergänzt wird, während der Nutzer gerade das Diagramm mit der Maus "abfährt". 

Ich bin mir noch nicht sicher, ob ich darauf Einfluss habe oder ob das ein Fehler in der verwendeten Chart-Library ist.

ledermann, 2023-09-14:
Mittlerweile (nach einigen Updates von Chart.js) scheint das Problem nicht mehr aufzutreten, zumindest kann ich es nicht mehr reproduzieren. Somit schließe ich das Issue.

closed 2023-09-14 by ledermann