solectrus/solectrus #1898: Show forecast for more days

Bart1909, 2023-04-07:
Hi @ledermann,

With a paid forecast api key you can get the forecast data for +3 or even +6 days.
The solectrus dashboard shows only the next day.

It would be nice, if more days are shown.

Best regards

Tobi

closed 2023-04-08 by ledermann

ledermann, 2023-04-08:
Good hint, that was easy to implement. Will be included in the next update.

ledermann, 2023-05-06:
Available now as part of version v0.10.1

Bart1909, 2023-05-06:
Working fine :)

[image]

noffycws, 2023-05-15:
Is there a documentation, how to integrate the forecast api key in Solectrus?

ledermann, 2023-05-15:
Hm, the [Forecast Collector](https://github.com/solectrus/forecast-collector) currently doesn't support specifying an API key at all. Of course, I could add this, but I'm curious how @Bart1909 did it.

Bart1909, 2023-05-15:
I modified the source code of the quarkus-forecast-collector and implemented the api key in the request url:
https://github.com/jbaesner/quarkus-forecast-collector/blob/d389342e573cbc7384b9d93e6f45294b690f08fd/src/main/java/org/solectrus/ForcastSolarApp.java#L64

I modified this line to:
`baseUrl = "https://api.forecast.solar/<myApiKey>";`

Bart1909, 2023-05-15:
Good to know: as mentioned [here](https://github.com/solectrus/forecast-collector/issues/64) I do not use the solectrus collector but instead use the [quarkus-forecast-collector](https://github.com/jbaesner/quarkus-forecast-collector/tree/main)

noffycws, 2023-05-15:
It is described here: https://doc.forecast.solar/api:estimate
There is an URL :apikey. With this key you can access to the paid functions.

@ledermann: Maybe you can add an variable FORECAST_APIKEY in the .env and use this in the forecast-collector.

ledermann, 2023-05-15:
Ok, thanks @Bart1909 and @noffycws for clarification.

I will add support for the API key in the near future, there is already an issue for this:
https://github.com/solectrus/forecast-collector/issues/72

commit e68464f7 (2023-04-08): Allow to paginate 6 days into the future

The forecasting data is limited to a maximum of 6 days.

Resolves #1898