solectrus/solectrus #1950: Erzeugung Peak-Wert (daily-peak)

Zeppelin-PV, 2023-04-26:
Es ist mitunter interessant den Peak-Wert seiner Anlage zu sehen was sie im Stande ist zu liefern.
Im "Jetzt" kann man diesen eine Stunde lang sehen, dann ist er weg.
In der Tagesansicht sieht man ihn nicht da die Anzeige 5min gerundet ist.
Könnte man den daily-peak irgendwie speichern und zur Anzeige bringen ?
... vielleicht in den Top Ten ?

ledermann, 2023-06-28:
Danke für diesen Vorschlag, den ich nun implementiert habe - als Ergänzung zur bereits bestehenden Top10 gibt es dort nun den "daily-peak" zu sehen. Auch steht dieser Peak für andere Messwerte bereit (Hausverbrauch, Akku-Beladung etc.)

Das Feature wird mit dem nächsten Update ausgerollt, bereits jetzt ist es in der Live-Demo zu sehen:
https://solectrus.ledermann.dev/top10/peak/inverter_power

Interessant in meinem Fall: Die Anlage hat laut Spezifikation 9,24 kWp, der tatsächliche Peak liegt mit 11,41 kWp erstaunliche 23 % darüber. Allerdings liegt dies bereits über zwei Jahre in der Vergangenheit - vermutlich Degradation oder Verschmutzung.

Zeppelin-PV, 2023-06-28:
außer bei der Erzeugung müssten doch alle anderen in der Peak Anzeige kW sein und nicht kWp
... oder täusche ich mich da

ledermann, 2023-06-28:
@Zeppelin-PV Streng genommen müsste wohl man *alle** Werte in kW angeben, also nicht in kWp. [Watt-Peak](https://de.wikipedia.org/wiki/Watt_Peak) ist ja ein etwas schwammiger Begriff, der eigentlich die **Nennleistung** unter bestimmten Standardtestbedingungen meint und sich nur auf Solarmodule bezieht.

Was ich hier in SOLECTRUS darstelle, ist aber keine Nennleistung, sondern eine tatsächlich erreichte maximale Leistung für einen bestimmten Messwert. Wenn die Überschrift (im Tab) schon "Peak" heißt, sollten das wohl in den einzelnen Balken nicht wiederholt werden. Insofern tendiere ich dazu, statt "kWp" besser "kW" zu schreiben.

Andere Meinungen dazu?

StevieC121176, 2023-06-28:
Ich denke auch, kW wäre besser

ledermann, 2023-06-29:
Ok, ich habe das auf `kW` abgeändert. 

Ich werde das Feature übrigens noch weiter ausbauen, damit man nicht nur den daily-peak sehen kann, sondern auch weekly-peak, monthly-peak und yearly-peak. Insbesondere letzteres ist nützlich, um die Degradation der Module zu beobachten. Stay tuned :-)

StevieC121176, 2023-07-01:
Peak für Tag/Woche/Monat/Jahr finde ich mit dem Umschalter sehr gut gelöst :-)

ledermann, 2023-07-01:
Ich habe nun, wie angekündigt, die Sache weiter ausgefeilt - siehe Live-Demo:
https://solectrus.ledermann.dev/top10/

Es gibt in der Navigation wieder nur Tag/Monat/Woche/Jahr wie zuvor, aber eine zusätzliche Auswahl zwischen Summe/Max für **jeden** der vier Zeiträume. Somit lässt sich nun auch z.B. der [Höchstwert nach Jahren](https://solectrus.ledermann.dev/top10/year/inverter_power/peak/desc) ermitteln. 

Womit ich noch nicht ganz zufrieden bin:

- Bei `Max` wird der absolut höchste Wert ermittelt, `Summe` entspricht dem bisherigen Verhalten. Das Maximum berechnet InfluxDB sehr performant (bei meinen Daten aus drei Jahren in 0,5 Sekunden). Das könnte aber auch ein einsamer Ausreißer sein. Sinnvoller wäre daher vielleicht ein Durchschnittswert für 5 Minuten - dann würde es nämlich zu den importierten SENEC-CSV-Daten passen. Allerdings würde dies erheblich an Performance kosten, d.h. die Anzeige wäre deutlich langsamer (bei mir: 4 Sekunden).
- Sind die Bezeichnung `Summe` und `Max` verständlich und gut gewählt? Gibt bessere Begriffe? Oder Icons? Ein "Σ" für Summe vielleicht?
- Die Bedienung des Selectors ist noch nicht perfekt (Farben, Maus, Mobile etc.)
- Ob die aufsteigende Sortierung so sinnvoll ist, sei mal dahingestellt

StevieC121176, 2023-07-01:
Mir persönlich würden Icons besser gefallen (Summen Zeichen wäre gut, Icon für Peak könnte man so wie im Beispiel unten darstellen)
Beim Selektor könnte man den selektierten Bereich an den Ecken evtl. auch abrunden wie die Aussenecken. Das wirkt dann eher wie ein Schalter.
Die ermittelten Peak Werte finde ich in Ordnung, ja das mögen Ausreißer sein, aber sie bilden den "Höchstwert" ganz gut ab.

ledermann, 2023-07-01:
Danke für das Feedback, ich werde das UI noch verbessern.

Bezüglich 5-Minuten-Durchschnitt: Üblicherweise dürfte es so sein, dass ein SENEC-User nicht vom ersten Tag an SOLECTRUS verwendet, sondern erst später darauf stößt und dann historische Daten importiert. Auch wenn dadurch der gesamte Zeitraum seit Installation der PV-Anlage abgedeckt ist, unterscheiden sich die Daten: **Vor** SOLECTRUS-Installation sind es 5-Minuten-Mittelwerte, **danach** 5-Sekunden-Einzelmessungen. Für die Summen ist das egal, aber für den Peak macht das einen erheblichen Unterschied. In den importierten Mittelwert-Daten finden sich nämlich nicht so hohe Peaks wie in den Einzelmessungen. 

Das könnte dann den Eindruck erwecken, als wäre die Anlage im Laufe der Zeit besser geworden, was aber natürlich nicht so ist. 

Konkret: Hier bei mir habe ich Messwerte im 5-Sekunden-Abstand nahezu von Beginn an, insgesamt seit knapp 3 Jahren. In der [Peak-Top10 nach Jahren](https://solectrus.ledermann.dev/top10/year/inverter_power/peak/desc) lässt sich eine jährliche Reduzierung von etwa 3 % erkennen (11,4 → 11,1 → 10,8), die mir als natürliche Degradation erscheint:

[image]

Das funktioniert aber nur, weil ich Einzelmessungen von Beginn an habe. Würde z.B. das erste Jahr nur aus importierten Daten bestehen, wäre der Peak im ersten Jahr erheblich kleiner (laut meinen Tests für 2021: 10,5 kW statt 11,4 kW).

Eine Lösung habe ich dafür noch nicht. Die schlechtere Query-Performance möchte ich vermeiden, auch wenn sich das durch aggressiveres Caching etwas abmildern lassen könnte. Ein zusätzliches Bedienelement im UI (z.B. "Max-peak" und "Max-gemittelt") möchte ich außerdem nicht, um die Bedienung nicht weiter zu verkomplizieren.

Naja, ich überlege dazu noch etwas weiter.

StevieC121176, 2023-07-01:
Ja das ist in der Tat etwas schwierig. Vielleicht könnte man in den Balken 2 Werte angeben, in Verschiedenen Farben. Schwarzer Wert ist der gemittelte und rot der absolute Spitzenwert, den evtl. noch in Klammer o.ä.

Bei Kostal verhält es sich ähnlich. Die exportierten Werte sind imho auch nur alle 5min. Den Peak Wert hab ich bis jetzt auch nur gehabt, da ich mir den über ioBroker in die influx Datenbank hab schreiben lassen. Jetzt mit Solectrus ist das hat. einfacher.

Zeppelin-PV, 2023-07-01:
ich würde das Wort Max durch Peak ersetzten
Peak drückt den "Spitzenwert" besser aus als Max, m.E.
die Schaltflächenanordnung ist gut so

ledermann, 2023-07-01:
Zu Euren Anmerkungen:

- Den Begriff "Peak" möchte ich in der Top10 eher nicht verwenden, weil das zu sehr mit dem Modul-Nennwert assoziiert wird. Hier geht es ja um die tatsächlich erreichten Werte.
- Beide Werte ("total-max" und "5-minute-max") zusammen möchte ich eher nicht anzeigen, das verkompliziert das Ganze, ohne viel Mehrwert zu bringen.

Ich habe es jetzt so gemacht:

- Es wird nicht mehr das absolute Maximum genommen, sondern das Maximum der gemittelten 5-Minuten-Intervalle, was natürlich eine etwas kleinere Zahl ergibt, aber eine Vergleichbarkeit von importierten und gemessenen Daten erlaubt. 
- Die Query von InfluxDB habe ich optimieren können, sodass es nicht mehr ganz so lange dauert. Das Ergebnis wird länger ge-cached.
- Statt der Texte Max/Sum werden Icons verwendet (von tabler: [math-max](https://tabler-icons.io/i/math-max), [sum](https://tabler-icons.io/i/sum)). Mit denen bin ich zwar nicht ganz glücklich, aber ich habe keine besseren gefunden. Außerdem gibt es Tooltips und das Styling habe ich angepasst, sodass es mehr nach einem Schalter aussieht.

Falls jemand SVG-Icons erstellen kann (oder bessere Icon-Bibliotheken zur Verfügung hat) - perfekt fände ich:

- Maximum: Stilisierte Kurve mit einem lokalen und einem globalen Maximum, bei der letzteres mit einem Punkt markiert ist
- Summe: Gleiche Kurve, bei der der Bereich unter der Kurve schraffiert ist

StevieC121176, 2023-07-01:
Gefällt mir sehr gut. Ladezeit für die Peak Werte ist ok finde ich.
Icons sind schwierig (obwohl deine verwendeten schon treffend sind), ich habe mal ein paar angehängt, die ich gefunden hab. 

[image]

[image]
[image]
[image]
[image]

StevieC121176, 2023-07-01:
was mir noch aufgefallen ist, in der Mobile Version ploppen die Tooltips für "aufsummierte Energie" und "maximale Leistung" für eine halbe Sekunde auf und verschwinden wieder. Ähnlich wie es mal für die Sortierung war.

ledermann, 2023-07-01:
Danke für Deine Icon-Vorschläge, auch wenn die mich nicht so recht überzeugen. Ich denke, ich lasse es erst mal so, wie es ist.

Die Tooltips habe ich jetzt für Touch-Displays deaktiviert, das fehlte tatsächlich noch.

closed 2023-07-01 by ledermann