solectrus/solectrus #1986: Kennzahlen in Kachel-Darstellung

ledermann, 2023-05-10:
Alternativ zur jetzigen Darstellung könnte es nützlich sein, eine weitere Seite zu haben, auf der ohne Diagramm nur einige wichtige Kennzahlen ersichtlich sind, beispielsweise:

- PV-Erzeugung Jetzt
- PV-Erzeugung Tag
- Akkustand Jetzt
- Temperatur Jetzt
- Ersparnis Tag

Die Darstellung könnte über Kacheln erfolgen. Einzelne Werte könnten als Tacho visualisiert werden.

StevieC121176, 2023-06-04:
Finde ich auch eine sehr gute Idee. Genial wäre das noch als Widget fürs Telefon. Ich habe mal im OpenWB Forum etwas gefunden. Diese Wallbox kann auch PV Monitoring. Dort hat ein User diese anpassbaren Widgets erstellt. Vielleicht könnte man so etwas ähnliches für Solectrus implementieren.
[image]
[image]

[image]
[image]

halmich, 2023-06-11:
gute Idee :)

StevieC121176, 2023-06-20:
Ich habe mich mal etwas mit den Widgets in IOS beschäftigt. So wie es scheint werden die Daten nicht aktuell dargestellt sondern die sind teilweise schon etliche Minuten alt. Das ist eher suboptimal.

von daher wäre eine Browserbasiertes Darstellung mit zeitnaher Darstellung der Livedaten besser. Ich habe mal ein Beispiel erstellt. Dieses ist Grafana basiert mit den Daten aus der influxdb
Vielleicht wäre sowas etwas für eine 2. Seite in Solectrus.
Das ist jetzt die mobile Darstellung, funktioniert hat. auch mit der Desktop Variante

[image]

ledermann, 2023-11-03:
Für dieses Issue hat sich mittlerweile ein Sponsor gefunden, der die Entwicklung finanziert :) 
Ich arbeite somit jetzt an der Implementierung, in der Live-Demo ist bereits etwas zu sehen. 

Die Kacheldarstellung ist zunächst allerdings fix, d.h. es gibt keinerlei Anpassungsmöglichkeit und es gibt auch keine integrierten Charts oder Tacho-Darstellungen. Das wären zukünftige Ausbaustufen.

StevieC121176, 2023-11-03:
Super. Auch das kleine Pulsieren des Wertes wenn er sich aktualisiert ist schick gemacht 👍 
Da bin ich schon gespannt wie das in Zukunft wird. 
Vielleicht könnte man die Anzeigewerte konfigurierter machen evtl. über eine Auswahlliste.
Solectrus wird immer besser.

ledermann, 2023-11-03:
Das Pulsieren war das Schwierigste ;-) 

Eine Konfiguration der Darstellung wird unvermeidlich sein, völlig klar. Jeder möchte da andere Dinge sehen. Aber das kommt später.

StevieC121176, 2023-11-03:
Unter IOS17 wird passt die Ansicht nicht ganz auf das Display. Evtl. könnte man die Kacheln  noch etwas in der Höhe kürzen das 6 Kacheln auf das Display passen?

[image]

ledermann, 2023-11-03:
Ja, ist bekannt, das Layout passt sich noch nicht ideal an die Höhe an. Kümmere ich mich noch drum. Ist alles noch im Fluss, der jetzige Stand ist nur der erste Wurf.

StevieC121176, 2023-11-05:
Cooles Icon für "Einsparung" 👍 :-)

closed 2023-11-07 by ledermann

ledermann, 2023-11-07:
Die Kacheln sind nun im develop-Image verfügbar. Feedback willkommen. Wenn jemand Fehler feststellt, bitte neues Issue anlegen. Es werden einige neuere Browser-Fähigkeiten genutzt ("Container Queries" und "Dynamic viewport units"), denkbar sind also Probleme mit älteren Browsern.

Zum Thema Customizing habe ich schon mal das Issue #2511 angelegt. Das wird aber vermutlich nicht kurzfristig realisiert werden - es sei denn, es findet sich ein weiterer Sponsor ;-)

Bart1909, 2023-12-04:
Moin @ledermann,
coole Ansicht, gefällt mir auch :)

Ich frag mich nur, was der Wert darstellen soll:
[image]

Es gibt keinen Hover-Text dazu und bei einem Klick darauf öffnet sich die Jahresansicht, wo ich den Wert aber nicht wiederfinde.

Vielleicht könnte man da noch eine Erklärung dran packen?

Viele Grüße

Tobi

ledermann, 2023-12-04:
@Bart1909 Das ist die CO₂-Einsparung in Tonnen, derzeitig nur auf der Einzelwerte-Ansicht dargestellt - und ja, hast recht, das muss noch besser herausgestellt werden. Habe #2594 angelegt.

Überhaupt wird sich diese Seite noch weiter verbessern, das ist nur eine erste Fassung :-)

PR #2498 (2023-11-07): Essentials view
Resolves #1986