solectrus/solectrus #2018: Make tooltips clickable on mobile UI

ledermann, 2023-05-22:
### Discussed in https://github.com/orgs/solectrus/discussions/2017

<div type='discussions-op-text'>

<sup>Originally posted by **GrimmiMeloni** May 22, 2023</sup>
The information in the tooltips is quite interesting and I like to take a look every now and then. On mobile however, the tooltips are basically not usable, as they rely on a non-existing hover event.

I'd like to suggest that the tooltips should be revealed by "clicking" event on mobile. I would consider it OK, to sacrifice the current click behavior on mobile, as these are merely shortcuts for the drop down selection (that is still available on mobile, too).</div>

Yes, this should definitely be implemented.

ledermann, 2023-05-24:
@GrimmiMeloni It seems to be a simple thing, but it was more complicated than expected. After a few tries, I have chosen this implementation:
 
As you know, the main screen consists of two sides: On the left is the balance, on the right a diagram. If you click an element on the left, the diagram is updated on the right. Up to now, however, this did not work as efficient as possible because both sides were updated, although nothing changed on the left.

This works differently now: A click on the left just updates the diagram on the right, nothing changes on the left. Now this allows a much better tooltip handling: They **remain** when clicked, which has exactly the desired effect on touch displays: A tap simultaneously displays the tooltip **and** executes the click event.

I also made some minor improvements to the click behavior:
- Disabled text selection (no more random text highlighting)
- Disabled touch callouts on iOS (no extra popup when clicking a link or button)
- Disabled dragging on iOS (no accidental moving of links or buttons)

See at https://solectrus.ledermann.dev

GrimmiMeloni, 2023-05-24:
Nice!
One little change I would suggest to make the new behavior on mobile more intuitive, would be to allow closing the tooltip by touching the element again. Right now, the only way I found to close the tooltip was to tap into the blank header space at the top.

ledermann, 2023-05-25:
@GrimmiMeloni You are right, I just fixed the closing. Now it works like this on touch displays: After a tooltip is opened, the next tap (no matter where) closes it.

GrimmiMeloni, 2023-05-25:
Just checked on ledermann.dev. This is excellent! 👌🏻

ledermann, 2023-05-26:
Fine, let's close this issue.

closed 2023-05-26 by ledermann

Bart1909, 2023-06-10:
Hi @ledermann,
when I now want to click on the balance to see the values of a value which is too low to be shown, it only updates the diagram but does not show the absolute value.
Before this change, I could see the tooltip (which wasn’t really nice working on mobile).
[image]
For example, I want to see, how much I consumed from battery today.

thanks in advance

Tobi