solectrus/solectrus #2032: Wrong values in hover for 'today' in 'week' and 'month' overview...

jbaesner, 2023-05-26:
When looking at the current data from _today_, e. g.:

[image]

The values are _in sync_ with what the SENEC App displays at that time, but switching to the _week_ or _month_ overview for _today_, the hover for today shows a wrong value, see:

**week:**
[image]

**month:**
[image]

This example shows the _production_, but it's the same for other types as well...

When using hovering past days, the values are correct.

ledermann, 2023-05-26:
You are right. The column in the diagram for the current day (or week/month/year) is **not correct**. However, this is only a temporary deviation. As soon as the day is over, it will be correct again. It's caused by the way InfluxDB handles incomplete data.

I already noticed this in the past, but I never wrote it down because the error always corrects itself. Thank you for finally doing it :-)

I'll see how I can fix that.

ledermann, 2023-05-29:
#1894 describes a similar problem. The reason for both issues is that the InfluxDB query used by SOLECTRUS can be inaccurate for non-closed time ranges. 

Tech details: I use this aggregation code, which has great performance. But, it doesn't work right when there's not a full hour of data. I can't find a function in InfluxDB that can fix this concern.

```
|> aggregateWindow(every: 1h, fn: mean)
|> aggregateWindow(every: 1d, fn: sum)
```

I have not found a solution to this issue yet, unfortunately. As soon as InfluxDB 3.0 is available, I'll take another look at the issue.

ledermann, 2024-07-03:
Ursache ist, dass mit Flux die Verwendung von `aggregateWindow(every: 1h, fn: mean)` ungenau ist, wenn der Zeitraum noch nicht abgeschlossen ist.

Lösungsansatz: 

Wenn das dargestellte Diagramm einen nicht abgeschlossenen Zeitraum betrifft (also aktuelle Woche/Monat/Jahr oder Gesamt), dann Sonderbehandlung mittels Split:

1. Zunächst abgeschlossenen Zeitraum wie bisher berechnen
2. Zusätzlich den laufenden (noch nicht abgeschlossenen Zeitraum) genau berechnen, über `integral(unit: 1h)`, mit [PowerSum](https://github.com/solectrus/solectrus/blob/develop/app/services/power_sum.rb)

Beispiel für die Berechnung des Diagramms für Juni 2024 am 14. Juni:

1. Diagramm für den Zeitraum vom 1. Juni bis 13. Juni wie bisher berechnen
2. Tag des 14. Juni mittels `PowerSum` berechnen

Das hat auch positive Auswirkungen auf die Performance, weil die Abfragen für abgeschlossene Zeiträume im Cache landen.

PR #3533 (2024-11-07): Auswertungen anhand vorberechneter Tageswerte
Dieser Pull-Request behebt verschiedene Probleme auf einmal:

- Keine Abweichungen mehr zwischen Diagrammen und Strombilanz (resolves #2032)
- Keine Abweichungen mehr zwischen Top10 und Strombilanz (resolves #1894)
- Keine Fehlberechnungen mehr bei Lücken in den Messwerten (wird immer wieder mal thematisiert, z.B. #2654)

Außerdem wird die Performance von Auswertungen (Strombilanz, Diagramme, Top10 und Kacheln) für längere Zeiträume deutlich verbessert. Die Abfragen sind um den Faktor 50 bis 100 schneller.

Wie diese Verbesserungen erreicht werden: [... cut]
closed 2024-11-07 by ledermann