solectrus/solectrus #2035: Top Ten

Zeppelin-PV, 2023-05-27:
in den Top Ten Rubrik Einspeisung wäre es schön wenn neben den kW noch der Einspeisebetrag eingeblendet würde.  z.B.  60kW - 4,82€
Auch in der Rubrik Netzbezug eine interessante Sache.  z.B.  10kW - 4,00€
Die Vergütung bzw. die Bezugskosten liegen ja in den Parametern,
LG an Georg, und in die Runde

ledermann, 2023-05-28:
Schöne Idee!

Zeppelin-PV, 2023-07-05:
nachdem nun in den Top-10 mit den Schaltflächen Summe und Peak differenziert werden kann könnte man doch das €-Zeichen nebenan platzieren und somit Erträge und Kosten anzeigen
Wehrmutstropfen; als reine finanzielle Anzeige würde der Bezug zu den KW's fehlen

ledermann, 2023-07-10:
Grundsätzlich stimme ich Dir absolut zu, @Zeppelin-PV. Bei Top10-Einspeisung könnte die zugehörige Einspeisevergütung, bei Top10-Netzbezug die zugehörigen Stromkosten aufgeführt werden. Natürlich jeweils nur bei Summe (in kWh), nicht bei Maximum (in kW).

Dass ich das nicht bereits gemacht habe, hat technische Gründe: Die Berechnung der Berträge ist nicht trivial. Je nachdem, ob man Tag, Monat, Woche oder Jahr gewählt hat, könnte es mehrere Preise je Balken geben. Bedenke, dass auch die Einspeisevergütung sich ändern kann, z.B. durch Erweiterung der PV-Anlage.

Letztendlich muss genau die gleiche Berechnung durchgeführt werden, die bereits in der Verlaufsansicht bei "Einsparung" gemacht wird. Diese müsste ich also extrahieren und für die Top10 verfügbar machen.

Ich werde das sicherlich bei Gelegenheit angehen. Bis dahin muss man auf den jeweiligen Balken der Top10 klicken, um in der Verlaufsansicht den Vertrag zu erfahren.