solectrus/solectrus #2036: Tage / Woche / Monat / Jahr / Gesamt - Error 500 interner Sever Fehler

NonBlackstation, 2023-05-28:
Ich habe erst seit zwei Tagen Solectrus in Benutzung und soweit ist alles super durchgelaufen.
Den Port 3000 nach aussen auf 3030 legen, da ich auf 3000 auch noch in einem anderen Container was da drauf habe.
Ich habe auch erfolgreich die csv Dateien von Senec importiert.

Wenn ich in die Übersichten Tag / Woche etc. möchte, erscheint der Fehler 500 : interner Server-Fehler. In den Logs von den Container finde ich nichts.
Was könnte hier schief gelaufen sein? 
Wo kann ich noch suchen?

Danke im Voraus

ledermann, 2023-05-28:
Bei einem Fehler 500 **MUSS** in den Logs etwas stehen, konkret im Log des App-Containers. Du kannst wie folgt dort nachsehen:

```bash
docker logs solectrus-app-1
```

(Name des Containers ggsf. anpassen, siehe `docker ps`)

Wenn der Output zu umfangreich und unübersichtlich ist, dann bitte folgendermaßen vorgehen:

```bash
docker logs solectrus-app-1 --follow
```

Anschließend im Browser auf die Seite von SOLECTRUS gehen, wo der Fehler auftritt (und ihn somit erneut auslösen) und dann in den Output gucken. Bitte den (relevanten) Output hier posten, dann kann ich weiterhelfen.

NonBlackstation, 2023-05-28:
2023-05-28T11:11:36.854282292Z | stdout | [172.18.0.1] ActionView::Template::Error (There is an issue connecting to your database with your username/password, username: postgres.
-- | -- | --
2023-05-28T11:11:36.854229205Z | stdout | F, [2023-05-28T11:11:36.854070 #12] FATAL -- : [172.18.0.1]
2023-05-28T11:11:36.852949181Z | stdout | I, [2023-05-28T11:11:36.852307 #12]  INFO -- : [172.18.0.1] method=GET path=/stats/inverter_power/2023-05-28 format=html controller=StatsController action=index status=500 duration=22.12 view=0.00 user_agent=Mozilla/5.0 (Macintosh; Intel Mac OS X 10_13_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/97.0.4692.99 Safari/537.36
2023-05-28T11:11:36.811850018Z | stdout | I, [2023-05-28T11:11:36.811677 #12]  INFO -- : [172.18.0.1] method=GET path=/500 format=html controller=ErrorsController action=show status=500 duration=2.60 view=2.11 user_agent=Mozilla/5.0 (Macintosh; Intel Mac OS X 10_13_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/97.0.4692.99 Safari/537.36
2023-05-28T11:11:36.809413651Z | stdout | [172.18.0.1] app/controllers/concerns/auto_locale.rb:14:in `switch_locale'
2023-05-28T11:11:36.809386196Z | stdout | [172.18.0.1] app/controllers/stats_controller.rb:5:in `index'
2023-05-28T11:11:36.809357444Z | stdout | [172.18.0.1] app/views/stats/index.turbo_stream.slim:11
2023-05-28T11:11:36.809330437Z | stdout | [172.18.0.1] app/controllers/stats_controller.rb:11:in `calculator'
2023-05-28T11:11:36.809302229Z | stdout | [172.18.0.1] app/controllers/stats_controller.rb:12:in `calculator'
2023-05-28T11:11:36.809274034Z | stdout | [172.18.0.1] app/controllers/stats_controller.rb:12:in `new'
2023-05-28T11:11:36.809243522Z | stdout | [172.18.0.1] app/services/calculator/range.rb:9:in `initialize'
2023-05-28T11:11:36.809213855Z | stdout | [172.18.0.1] app/services/power_sum.rb:8:in `call'
2023-05-28T11:11:36.809185568Z | stdout | [172.18.0.1] app/services/power_sum.rb:32:in `sum'
2023-05-28T11:11:36.809156565Z | stdout | [172.18.0.1] app/services/power_sum.rb:50:in `price_sections'
2023-05-28T11:11:36.809124061Z | stdout | [172.18.0.1] app/services/date_interval.rb:12:in `price_sections'
2023-05-28T11:11:36.809090477Z | stdout | [172.18.0.1] app/services/date_interval.rb:42:in `prices'
2023-05-28T11:11:36.809065753Z | stdout | [172.18.0.1]
2023-05-28T11:11:36.809029251Z | stdout | [172.18.0.1]     14:   turbo-stream action="update" target="chart"
2023-05-28T11:11:36.809000434Z | stdout | [172.18.0.1]     13: - if params[:chart] != 'false'
2023-05-28T11:11:36.808972253Z | stdout | [172.18.0.1]     12:
2023-05-28T11:11:36.808941140Z | stdout | [172.18.0.1]     11:       = render DashboardRange::Component.new calculator:, timeframe:
2023-05-28T11:11:36.808913599Z | stdout | [172.18.0.1]     10:     - else
2023-05-28T11:11:36.808881107Z | stdout | [172.18.0.1]      9:       = render DashboardNow::Component.new calculator:
2023-05-28T11:11:36.808852161Z | stdout | [172.18.0.1]      8:     - if timeframe.now?
2023-05-28T11:11:36.808826818Z | stdout | ):
2023-05-28T11:11:36.808785106Z | stdout | Please check your database configuration to ensure the username/password are valid.
2023-05-28T11:11:36.808757701Z | stdout |  
2023-05-28T11:11:36.808670399Z | stdout | [172.18.0.1] ActionView::Template::Error (There is an issue connecting to your database with your username/password, username: postgres.

Hmmmm, bitte entschuldigen Sie meine Aussage. Ich habe in den Logs es wohl übersehen. Wenn es natürlich keinen Zugriff zu Datenbank gibt, dann kann die Historie auch nicht angezeigt werden.
Wie kann ich dies lösen? ich habe noch einen weiteren Container am laufen -solaranzeige- wo ich auch eine influxdb auf 8086 habe.

NonBlackstation, 2023-05-28:
Protokoll von solectrus_db_1
<div class="x-grid3-row x-grid3-row-selected" role="row" id="ext-gen6329" style="margin: 0px; padding: 0px; cursor: default; border-width: 0px 0px 1px; border-style: dotted dotted solid; border-top-color: initial; border-right-color: initial; border-bottom-color: rgba(198, 212, 224, 0.4); border-left-color: initial; border-image: initial; width: 1353px; background-color: rgba(5, 127, 235, 0.1) !important; color: rgb(65, 75, 85); font-size: 13px; font-family: Verdana, Arial, sans-serif; font-style: normal; font-variant-ligatures: none; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial; height: 24px;">

2023-05-28 11:19:56.200 UTC [14427] FATAL:  password authentication failed for user "postgres" | stderr
-- | --

</div>

ledermann, 2023-05-28:
Ok, es handelt sich hier um ein Problem mit der Datenbank PostgreSQL, also **nicht** um InfluxDB. In PostgreSQL werden die Strompreise abgelegt, in InfluxDB hingegen die Messwerte.

Um das Problem zu beheben: In der `.env`-Datei muss ein Eintrag zu `POSTGRES_PASSWORD` stehen, der in der `docker-compose.yml` zweimal referenziert wird (bei den services `db` und `app`). Dadurch wird die Datenbank mit dem Passwort beim ersten Start eingerichtet und die App weiß es auch, um darauf zuzugreifen.

Dieses Passwort darf im Nachhinein (also bei bestehender Datenbank) nicht mehr geändert werden, vermutlich ist das aber passiert. Die Korrektur wäre also, das zuvor dort eingestellte Passwort wieder einzustellen und die Container dann neu zu starten. Notfalls kann man aber auch den `postgresql/`-Ordner löschen, der wird beim nächsten Start dann neu angelegt (man verliert dann nur die bereits dort eingestellten Strompreise).

Ich hoffe, das hilft. Ansonsten bitte nochmal melden :-)

NonBlackstation, 2023-05-28:
Vielen Dank Herr Ledermann !!!!
Die nun verwendeten Ports sind 3030:3000 und 8087:8086, des weiteren habe ich wirklich mal ein Passwort  und da kam "zu kurz" oder so ähnlich. Ich habe den postgresql-Ordner gelöscht, die csv neu importiert und nun funktioniert es.
Bitte entschuldigen Sie, ich den Fehler vor lauter logs nicht mehr gesehen :)

Ich wünsche noch angenehme Tage

ledermann, 2023-05-28:
Kein Problem - freut mich, dass es nun klappt. Ich wünsche viel Freude mit SOLECTRUS 😄

closed 2023-05-28 by ledermann