solectrus/solectrus #2056: Feature request: Gesamtdiagramm bei Jetzt und Tagesansicht

StevieC121176, 2023-06-04:
Ich würde mir in der Jetzt und in der Tagesansicht ein Gesamtdiagramm mit mehreren relevanten Daten wie Erzeugung, Hausverbrauch, Einspeisung usw.
Zur Verdeutlichung habe ich mal ein Beispiel angehängt.

[image]

[image]

ledermann, 2023-06-05:
Ich bin zwar kein Freund dieser Darstellung, weil ich es als Information-Overkill empfinde, aber ich lasse das gerne mal hier stehen und warte ab, ob das ein auch von anderen gewünschtes Feature ist.

halmich, 2023-06-11:
Im obigen Diagramm ist tatsächlich die volle Dröhnung. Allerdings finde ich eine Übersicht im obigen Stil, welche gleichzeitig die Erzeugung und den Hausverbrauch, und vielleicht noch den Netzbezug anzeigt gar nicht schlecht.

Deanw1975, 2023-06-23:
Hallo zusammen, schließe mich an. Es sollte so sein, dass jeder entscheiden kann welche Infos übereinander gelegt werden können. Dann wird es auch kein Overkill;-)
LG
Dean

Robbe64, 2023-10-04:
Ich fände das auch sehr gut! :)

Thalkirchner, 2023-10-05:
Keine schlechte Idee, aber nur wenn Inhalte auswählbar, wie bei der SENEC App

AlpenFlizzer, 2024-06-03:
Ich hole das hervor, da ich so eine Ansicht auch sehr mag.

Solaredge löst das in deren App so:
[image]

Das finde ich nicht so sehr overkill, gibt aber trotzdem eine gute Sicht, wie wirtschaftlich der PV Strom angelegt wird. 
Da ich leider keine Batterie habe (war bei österreichischen Strompreisen/Einspeisevergütungen nie wirklich relevant - bis jetzt vielleicht...) reicht mir der abgespeckte Graph auch.

Ich habe jetzt eine Entwicklungsumgebung für Solectrus am Laufen und möchte dem mal einen Versuch widmen... Disclaimer: bin zwar IT-Mensch aber kein professioneller Dev, daher ist das ergebnisoffen...

AlpenFlizzer, 2024-06-15:
Kleines Update zum Zwischenstand, so schaut es aktuell aus:
[image]

Ich nähere mich zumindest schon meiner SolarEdge Grafik an - aber es gibt noch einiges zu tun und zu checken... Graidenten und Transparenz sind nicht optimal. Andere oder mehrere Sensoren übereinanderlegbar? Sind die Werte überhaupt korrekt usw...

StevieC121176, 2024-06-16:
schaut aber schon sehr gut aus.

ledermann, 2024-06-16:
@AlpenFlizzer: Tolle Sache - bin ganz fasziniert davon, wie du in der Lage bist, meine bisherige Arbeit weiter auszubauen! 

Hier nur ein paar kleine Anmerkungen:

- Statt Hausverbrauch und Netzbezug wäre es vielleicht interessanter, Verbrauch (= Summe über alle Verbraucher) und Eigenverbrauch (= Anteil des Verbrauchs an der Erzeugung) darzustellen. Also wirklich so, wie es SolarEdge macht. 
- Bei größeren Timeframes (ab Woche) könnte man vielleicht mit gestapelten oder nebeneinander liegenden Balken arbeiten.
- Ein Customizing (= User kann selbst wählen, welche Graphen er übereinander legen will) würde ich zumindest in der ersten Fassung weglassen, um es einfach zu halten. Die "Kombinierte Ansicht" wäre also ein fest vorgegebenes Diagramm.

ledermann, 2024-06-18:
Als Verbrauch hätte ich momentan nur `house_power + heatpump_power + wallbox_power` angesehen, aber ich bin da nicht so tief im Thema drin.

Kann ich denn irgendwo sehen, was genau du versuchst hast? In [deinem Repo](https://github.com/AlpenFlizzer/solectrus) sehe ich noch nichts dazu.

Eine Custom-Flux-Query würde ich auch eher vermeiden, das müsste eigentlich mit der bestehenden `PowerChart`-Klasse funktionieren. Ein Problem aber:

Das einzige Diagramm, das derzeitig zwei Kurven enthält, ist Erzeugung+Vorhersage in [ChartData#data_day_inverter_power](https://github.com/solectrus/solectrus/blob/develop/app/services/chart_data.rb#L37-L54). Dieses arbeitet ungünstigerweise mit ZWEI Queries, was eigentlich nicht nötig wäre. Das sollte auch mit einer gehen. 

Da du jetzt noch mehr Kurven zeichnen möchtest, sollten die Werte möglichst auch alle in einer Query abgerufen werde. Letztendlich müssen in der Klasse `ChartData` ein paar Dinge umgebaut werden, denke ich. Und die Berechnung, also dass einzelne Werte addiert werden, muss auch ergänzt werden.

ledermann, 2024-06-18:
Ich habe das erwähnte Problem der unnötigen zwei Queries gerade [behoben](https://github.com/solectrus/solectrus/commit/895139592788ea4499def69b3c8496ce48e510bf). Der gleiche Ansatz könnte auch dir helfen, grundsätzlich mehrere Kurven zu behandeln. Was bleibt, ist natürlich die Durchführung der Berechnung, also z.B. das Addieren.

AlpenFlizzer, 2024-06-18:
Ich habe die Berechnungen nun doch endlich hinbekommen. Ich muss nochmal ordentlich aufräumen und danach merge ich den feature mal in meinen develop.

Habe jetzt wie gesagt in einem Chart 
Produktion (`inverter_power`), 
Verbrauch (`house_power+heatpump_power+wallbox_power`) Mein Gehrin mag nicht kapieren, ob die Batterie Beladung auch addiert werden muss... Habt ihr dazu noch Meinungen?
und Eigenverbrauch (`inverter_power-grid_export_power`) https://energis.de/ratgeber/photovoltaik/photovoltaik_eigenverbrauch_berechnen

Es deckt sich grundsätzlich mit meinem SolarEdge Chart, aber mir ist jetzt aufgefallen, dass die Daten minimal abweichen. Das liegt daran, dass ich am Hauszähler an der P1 Schnittstelle Import und Export ablese und die Produktion vom Wechselrichter. SolarEdge bezieht alle drei Daten natürlich vom SolarEdge Zähler, der mit Strommesszangen nach dem Zähler an den Phasen misst. Der Unterschied kommt also daher und den muss ich in Kauf nehmen...

Sieht für einen Tag und größeren Zeitrasum aktuell so aus:
[image]

[image]

ledermann, 2024-06-20:
Sieht wirklich großartig aus 👍 Bin gespannt auf den Code!

closed 2026-01-19 by ledermann