solectrus/solectrus #2059: Anzeige Akkukapazität

StevieC121176, 2023-06-05:
Wäre es möglich die aktuelle Akkukapazität durch z. Bsp. einen Tooltip anzuzeigen.

Beispiel Mauszeiger auf die % Anzeige Batterie -> 11,52 von 12,8kWh bei 90% Akku

 
[image]
[image]

ledermann, 2023-06-05:
Ja, wäre prinzipiell machbar. Derzeitig kennt SOLECTRUS die Kapazität des Speichers jedoch nicht, d.h. die müsste erst noch erfasst werden (am besten per `.env`).

StevieC121176, 2023-06-05:
EVCC stellt diesen Wert per MQTT zur Verfügung. Ich denke auch es wäre analog der anderen Werte in der *.env am besten aufgehoben.