solectrus/solectrus #2064: Mobile Design

StevieC121176, 2023-06-06:
Hi. Wenn bei mir in der unteren Zeile das JETZT oder das GESAMT markiert ist, wird es durch das abgerundete Display des iPhones abgeschnitten. Sieht man auf dem Screenshot jetzt leider nicht.
Könnte man die oberen beiden Zeilen nicht zu einer zusammenfassen. Diese nehmen schon einen Großteil des Bildschirms ein bei verhältnismäßig wenig Informationen.
Oder die Auswahl JETZT, Tag... wie in der DesktopAnsicht mit in die obere 2. Zeile packen?

[image]

[image]

ledermann, 2023-06-06:
Danke für Deine Anregung, aber mir ist noch nicht ganz klar, was genau Du meinst. Hier ein Screenshot für ein iPhone 14 inklusive Rahmen, angefertigt mit dem iOS-Simulator von Xcode:

[image]

So sieht es bei mir auch in Natura aus und so soll es auch aussehen.

Ja, die linke untere Ecke mit dem Button "Jetzt" wird leicht vom Display-Rand überlagert und somit angeschnitten. Ein Problem resultiert daraus aber nicht, soweit ich das sehe. Der Button ist weiterhin funktionsfähig, ich teste das auf verschiedenen iPhones.

Ein Zusammenfassen der oberen beiden Zeilen könnte eine Möglichkeit sein, Platz zu sparen. Das müsste ich mal in Ruhe evaluieren. Die Bottom-Navigation auf dem Smartphone möchte ich hingegen gerne belassen (zumindest von der Position her), weil das ein gewohntes Navigationselement ist. Deren Darstellung könnte aber verbessert werden (z.B. Klickbereiche der Buttons vergrößern).

ledermann, 2023-06-06:
Das mit dem Blockly-Script klingt gut 👍 

Für das UI habe ich jetzt ein paar kleine und einfache Verbesserungen vorgenommen:

- Zeilenhöhe 1 und 2 etwas verkleinert (nur auf Smartphone)
- BottomNavigation (nur für Smartphones) anders gestaltet, es sind jetzt eher Tabs als Buttons. Der klickbare Bereich ist dadurch größer und es sieht nichts mehr "abgeschnitten" aus

Auf dem iPhone sieht das jetzt so aus:

[image]

Live auch auf https://solectrus.ledermann.dev zu sehen.

StevieC121176, 2023-06-06:
Im Hochformat sieht das sehr gut aus, im Querformat müssten die Tabs etwas breiter sein, da klickt man leicht zwischen den Tabs ins Leere.

ledermann, 2023-06-06:
Stimmt, da fehlte noch was. Ich habe jetzt nochmal nachgelegt, auch für andere Bildschirmgrößen. Jetzt passt es.

iPhone `landscape` ist dennoch nicht optimal. Da bin ich mir noch unschlüssig, ob ich das vielleicht ganz deaktivieren sollte (also `portrait` erzwingen). Oder ob ich bei `landscape` Strombilanz und Chart nebeneinander quetsche. Auf jeden Fall ist es ungünstig, wenn man scrollen muss. Ich selbst nutze SOLECTRUS auf dem iPhone auschließlich hochkant.

StevieC121176, 2023-06-06:
Ja so passt es. Querformat ist ungünstig mit dem nach scrollen. Vielleicht wäre Strombilanz und Chart nebeneinander ein Kompromiss. Ich nutze auf dem iPhone auch nur den Portrait Modus. Auf dem iPad allerdings den Landscape.

StevieC121176, 2023-06-07:
Dieser dezente Farbverlauf in der Livedemo gefällt mir sehr gut.

Zum Thema Darstellung ist mir etwas auf einem Tablet aufgefallen. Ich habe noch ein Samsung Tablet hier, welches ich noch als Monitor für Solectrus und EVCC nutzen möchte. Lasse ich jetzt Solectrus normal über den Browser laufen fehlen diese Kreisdarstellungen für Autarkie, Verbrauch, Temperatur und Akku. Lasse ich Solectrus über Samsung Des laufen (was ja einer Desktopansicht entspricht) werden diese Kreisdarstellungen angezeigt. Wenn ich beide Vergleiche könnte man doch In der Landscape Variante auf dem Tablett diese Kreisdarstellung generell anzeigen?
Zum Vergleich hier mal beide Darstellungen auf meinem Tablet:

<details>
 <summary>Screenshots</summary>

[image]

[image]

</details>

ledermann, 2023-06-07:
Grundsätzlich ist es so, dass je nach verfügbarem Platz einige Bedienelemente ein- bzw. ausgeblendet werden oder auch (wie im Falle der Badges) anders aussehen. Die Badges erhalten einen kreisförmige Umrandung (die mehr Platz benötigt), wenn das Browserfenster in der Breite 1280px oder größer ist.

"Samsung Des" sagt mir nichts - ich bin allerdings in der Android-Welt auch nicht zuhause. Deine Screenshots wirken aber, als würden sie unterschiedliche Auflösungen verwenden. Auf dem zweiten Screenshot ist die Schrift kleiner, vermutlich wegen einer höheren Auflösung und somit erscheinen die Kreise.

Die Lösung für Deinen Fall könnte also sein, die Auflösung zu erhöhen (sofern das möglich ist).

StevieC121176, 2023-06-08:
So genau kenne ich mich mit Android auch nicht aus, nutze auch Apple. Ich weiss nur Samsung DEX ist so eine Art Desktop für Android. Schon möglich das dadurch eine andere Auflösung genutzt wird. Einstellen kann man da nichts soweit mir bekannt.

spirit0607, 2023-06-08:
> Ja genau das meinte ich mit der überlagerten Ecke. Ist nur ein optisches Problem. Funktion ist dadurch nicht eingeschränkt.
> 
> 
> 
> Status ist bei mir jetzt auch nicht mehr nur "Live" und "Fail" Ich schreibe jetzt meine Wechselrichter Statusmeldungen per Blockly Script in einen Datenpunkt der durch MQTT published wird und ich mir den dann in Solectrus anzeigen lasse.
> 
> 
> 
> Vielleicht könnte man die oberen 2 Zeilen etwas in der Höhe minimieren, das würde auch noch etwas Platz schaffen.
> 
> 
> 
> 

Könntest du mir das Blockly Script zur Verfügung stellen? Oder könnte man das auch evtl. mit in die Installationsanleitung MQTT als "optional" mit aufnehmen?? Das wäre super. Danke!

ledermann, 2023-06-08:
Für Dein Samsung-Tablet habe ich keine weitere Idee. Offenbar wird eine zu kleine Auflösung (< 1280px) verwendet, so dass die Kreise nicht dargestellt werden. Die Darstellung von SOLECTRUS ist aber korrekt.

Zwischenzeitlich (siehe oben) habe ich die Darstellung für Smartphone-portrait optimiert, für Smartphone-landscape ist es weiterhin nicht optimal, weil man scrollen muss. Die Darstellung von Strombilanz und Chart **nebeneinander** habe ich getestet, das haut nicht hin - zu wenig Platz. Denkbar wäre, in diesem Szenario das Chart ganz wegzulassen, aber das ist alles eine knifflige Sache, weil schon die Strombilanz alleine gewisse Höhe benötigt, um sinnvoll dargestellt zu werden. Ich lagere das mal in ein separates Issue aus, das ich später mal angehe. Für's erste denke ich, kann dieses Issue geschlossen werden.

closed 2023-06-08 by ledermann

StevieC121176, 2023-06-08:
Kein Problem mit dem Samsung Tablet, das ist schon etwas älter. Und in diesem DEX Modus scheint die Auflösung höher zu sein.

Ich muss sagen die Anpassungen (verkleinerte Kopfzeile, Tabs unten und die etwas andere Platzaufteilung gefallen mir sehr gut. Gibt es so eine Art Nightly oder ist so etwas geplant? Oder sind solche Änderungen erstmal nur in der Livedemo?

StevieC121176, 2023-07-02:
Ich schreib es mal hier rein, da ich dafür kein neues Issue aufmachen wollte.
In der mobile Ansicht wird die Auswahl für "Erzeugung", "Hausverbrauch" usw jetzt als Button dargestellt. Die Schrift hat sich auch vergrößert (Ist auf der Seite jetzt am größten m.M.n.) . Das sieht dadurch etwas "übermächtig" aus. Wäre es möglich die Schrift in der mobile Version wieder etwas zu verkleinern das der Button nicht ganz so groß ist?

[image]

ledermann, 2023-07-02:
Danke für den Hinweis, dieser Button war unbeabsichtigt, ich habe es korrigiert.

Momentan schraube ich am UI, was manchmal unerwünschte Nebeneffekte hat. Das sind die Nachteile des `develop`-Branches - man bekommt auch jeden kleinen Fehler mit ;-)