solectrus/solectrus #2084: Tooltips for balance segments not visible on touch display

ledermann, 2023-06-12:
@Bart1990 wrote in #2018:

> Hi @ledermann,
> when I now want to click on the balance to see the values of a value which is too low to be shown, it only updates the diagram but does not show the absolute value.
> Before this change, I could see the tooltip (which wasn’t really nice working on mobile).
> [image]
> For example, I want to see, how much I consumed from battery today.
>
> thanks in advance
>
> Tobi

_Originally posted by @Bart1909 in https://github.com/solectrus/solectrus/issues/2018#issuecomment-1585810969_

StevieC121176, 2023-06-20:
Was mir im Zusammenhang mit den Tooltips in der mobilen Touch Version noch aufgefallen ist:
Wenn ich in den TopTen auf Sortierung klicke wird für einen Bruchteil einer Sekunde ein Tooltip eingeblendet „Sortierung wechseln“ da es so kurz eingeblendet wird sieht es etwas unschön aus. 
Mich weiß nicht ob man das bei iOS abfragen kann, aber da gibt es doch verschiedene druckpunkte um Aktionen auszuführen. Leichter Druck - Tooltip wird angezeigt, stärkerer Druck -  Aktion wird ausgeführt.

ledermann, 2023-06-21:
Eine sinnvolle Tooltip-Behandlung ist gar nicht so trivial... Ich habe hier jetzt für Touch-Displays verbessert:

- Die Tooltips in den einzelnen Bilanzsegmenten werden wieder angezeigt (aber erst bei einem "press-and-hold")
- Der Tooltip auf dem Sort-Toggle in der Top10 wird nicht mehr angezeigt, da er kaum Mehrwert liefert

Auf Desktop-Display ist alles unverändert.

Die Änderung fließt ins nächste Update mit ein. Zu sehen ist es aber bereits jetzt in der [Live-Demo](https://solectrus.ledermann.dev).

StevieC121176, 2023-06-21:
Es wird auch bei einem langen Druck das Diagramm aktualisiert , wäre es möglich das Diagramm nur bei einem kurzen Druck zu ändern? Und beim langen Druck nur den Tooltip anzuzeigen?

Zeppelin-PV, 2023-06-22:
Bei meinen mobilen Geräten (Nokia 7.2 Firefox und Chrome And.11, Lenovo Tab K11 Firefox und Chrome And.12) sieht es so aus
nach "press and hold"
[image]
erst wenn ich das Fenster wegklicke erscheint das Tooltip
[image]
Das war aber auch schon vor der Deaktivierung auf mobilen Geräten so.

Zeppelin-PV, 2023-06-22:
Ergänzung:
mit zwei Fingertip "press n' hold" funktioniert es solange die Finger den Touch berühren
Das Tooltip verschwindet sobald die finger den Touch verlassen.
Bei kleinen Fenstern (Verbrauch 0,4kW siehe oben) mit zwei Fingern auf 6,1Zoll (Nokia 7.2)  - eher unpraktisch

Zeppelin-PV, 2023-06-22:
toller Nebeneffekt:
tippe ich mit drei Fingernauf drei verschiedene Felder werden alle 3 Tooltips angezeigt (Lenovo Tab)
[image]
funktioniert auch mit 4 Fingern

ledermann, 2023-06-22:
Danke für Euer Feedback. Ich habe jetzt noch eine Änderung ergänzt, sodass auch auf Android-Geräten die Link-Popup-Anzeige nicht mehr erscheint (bei iOS war das schon immer der Fall). Somit sollte jetzt ein normaler LongPress funktionieren. Das ist allerdings ungetestet, da ich gerade kein Android zur Hand habe.

Was ich erst mal nicht weiterverfolge, weil ich dafür keine (schnelle) Lösung finde:

- Click-Event wird nach LongPress ausgeführt, d.h. das Diagramm wird neu geladen, obwohl man eigentlich nur den ToolTip sehen möchte
- Gleichzeitige Darstellung verschiedener Tooltips, wenn man mit an verschiedenen Orten gleichzeitig tippt

Beides hat meines Erachtens keine sonderlichen Nachteile, dafür möchte ich keine weitere Energie verwenden.

closed 2023-06-22 by ledermann

Zeppelin-PV, 2023-06-22:
... es hat sich bei den Androiden nichts geändert, ist mir persönlich auch nicht sooo wichtig
wollte zu diesem Thema nur mein Feedback geben, hab' momentan nicht die Mittel für iOS Geräte ;-)
[image]

StevieC121176, 2023-06-22:
Das Neuladen des Diagramms kann man verhindern, wenn man während des LongPress den Finger leicht nach oben oder unten zieht. Da reichen einige Millimeter. Dann wird der Tooltip angezeigt und beim Loslassen das Diagramm aber nicht neu geladen.

Zeppelin-PV, 2023-06-22:
Super Tipp !!!
so popt auch das Link-Fenster nicht auf.

Bart1909, 2023-06-26:
Hi @ledermann ,
hast du vor, dafür einen Fix rauszubringen? 
Oder wartest du, bis du eine neue Minor-Version rausbringst?

Schöne Grüße aus der Nähe
Tobi

ledermann, 2023-06-27:
@Bart1909 Ja, es wird auf eine neue Minor-Version ([v0.12.0](https://github.com/solectrus/solectrus/milestone/14?closed=1)) hinauslaufen. Bugfix-Updates möchte ich nur machen, wenn es um wirklich dringende Sachen geht.

Wer am aktuellen Stand der Entwicklung teilhaben möchte, kann das `develop`-Image verwenden, das täglich mehrfach aktualisiert wird:
`ghcr.io/solectrus/solectrus:develop`

Gruß aus Jülich! :)