solectrus/solectrus #2094: Anpassung Diagrammbreite an Fensterbreite Desktop

StevieC121176, 2023-06-18:
Wie im Video unten zu sehen passt sich Solectrus ja schön an die Fenstergröße an, allerdings nur bis zu einem gewissen Punkt. Das Diagramm ist dadurch in der Breite recht "gequetscht" Hat man die volle Fensterbreite gibt es rechts und links schon reichlich freie Fläche. Könnte man da nicht das Diagramm in der X-Achse noch etwas breiter ziehen. In der mobile Version sieht das Diagramm "harmonischer" aus, da ist das Verhältnis X/Y Achse stimmiger.

<details>
 <summary>Video</summary>

https://github.com/solectrus/solectrus/assets/113791299/d928e1b0-af0c-47f4-999f-eb79ee811c19

</details>

ledermann, 2023-06-18:
Danke für den Hinweis! Es ist tatsächlich so, dass nicht die gesamte zur Verfügung stehende Breite genutzt wird - weil das bei sehr großen Breiten auch nicht optimal wäre. 

Ich habe jetzt die maximale Breite um 20% erhöht, d.h. bei sehr großen Screens (ab 1920px in der Horizontalen) ist die Darstellung nicht mehr 1280px breit, sondern 1536px. In der Live-Demo (https://solectrus.ledermann.dev) ist das bereits zu sehen.

closed 2023-06-18 by ledermann

StevieC121176, 2023-06-18:
Hab mir das grad in der Livedemo angeschaut. So passt das besser. Super.