solectrus/solectrus #2095: Anzeige prozentuale Verteilung Hausverbrauch (Netz/Batterie/PV) , Erzeugung (Einspeisung/Verbrauch)

StevieC121176, 2023-06-18:
Ich finde diese Verteilung immer ganz spannend, vielleicht könnte so eine Anzeige als Auswahl mit in die einzelnen Sichten (Jetzt, Tag, Woche...) mit als Auswahl aufgenommen werden

<details>
 <summary>Beispiel</summary>

[image]
[image]

</details>

ledermann, 2023-06-18:
Ja, gute Idee. Die Darstellung könnte im Tooltip erfolgen (für Erzeugung und Haus und auch Wallbox).

StevieC121176, 2023-06-20:
Ich habe mir das Charts.js mal angesehen. Evtl. könnte man diese Darstellung auch in eine Art Donau Diagramm anzeigen. Darin kommt eine Prozentuale Darstellung gut rüber wie ich finde.

[image]

ledermann, 2024-08-28:
Im Wesentlichen ist das [durch den Power-Splitter bereits abgedeckt](https://github.com/solectrus/solectrus/issues/2436), der Tooltip bei den Verbrauchern ist implementiert.

Bei zwei Details weicht die jetzige Implementierung (zu sehen in der [Live-Demo](https://demo.solectrus.de/inverter_power/2024-08)) von der oben skizzierten Idee ab:

1) Bei Erzeugung gibt es keine Prozentangabe. Es gibt aber schon lange den Badge "Verbrauch" mit genau der gewünschten Angabe.
2) Bei den Verbrauchern wird nicht errechnet, wie viel aus der Batterie kommt. Es wird nur zwischen PV und Netz unterschieden. Wenn also Strom aus der Batterie kommt, wird das als PV gewertet.

Sind diese beiden Punkte weiterhin gewünscht? Aus meiner Sicht würde kein großer Mehrwert entstehen. Oder seht ihr das anders?

AlpenFlizzer, 2024-08-28:
Ich hatte das in der Live Demo nicht gesehen. Ich wäre damit zufrieden, was auch daran liegt, dass bei mir kein Speicher im Einsatz ist. Das Proble ist aber, dass man ohne Verwendung des Power Splitters die Aufteilung nicht bekommt oder? Also bei mir wird im Tooltip für Hausverbrauch (habe keinen Speicher, Wärmepumpe oder Batterie) nichts angezeigt.
[image]

ledermann, 2024-08-28:
@StevieC121176 Der Gedanke beim Power-Splitter ist bislang, dass Strom, der aus dem Speicher kommt, zuvor höchstwahrscheinlich selbst erzeugt worden ist. Auch wenn durch Ladeverluste einiges verloren geht, sehe ich das als PV-Strom. Wichtig ist mir, wie viel aus dem Netz bezogen werden musste, dann dafür fällt ein anderer Preis an.

Eine rechnerische Ermittlung, wie viel Strom des Verbrauchers aus dem Speicher kommt, wäre prinzipiell zwar auf die gleiche Weise denkbar - aber es gibt Detailprobleme: Ich kann beispielsweise meinen SENEC-Speicher (mit integrierter Wallbox-Steuerung) so einstellen, dass auch das Auto aus dem Speicher beladen wird. An den Verbrauchswerten ist das aber nicht zu erkennen. Im Power-Splitter kann ich somit nicht eindeutig ermitteln, ob eine Akkuentnahme ins Haus oder ins Auto geflossen ist. Aber vielleicht habe ich das noch nicht genügend durchdacht. Eventuell könnte das später mal ergänzt werden. Ich benenne dieses Issue mal um, weil die Ausweisung des Batterie-Anteils ist das einzige, was der Power-Splitter noch nicht kann.

Die Ersparnisberechnung läuft derzeitig unabhängig vom Power-Splitter, da wird die gesamte Anlage betrachtet. 

@AlpenFlizzer Ja, die Aufteilung erfolgt nur durch den Power-Splitter. Dieser muss parallel laufen (in einem eigenen Docker-Container) und berechnet permanent die Verteilung des Netzstroms auf die Verbraucher. Die Dashboard-App guckt nach, ob in der InfluxDB Power-Splitter-Werte zu finden sind. Falls ja, werden diese angezeigt, andernfalls sieht alles aus wie bisher.

Aktuell läuft in #3330 noch der Betatest - da steht auch genau, wie man den zusätzlichen Container einrichtet. Ich plane, die Testphase sehr bald abzuschließen und das Feature mit Version 0.16 offiziell zu machen.

AlpenFlizzer, 2024-08-28:
@ledermann ich habe es schlecht beschrieben, aber der Power Splitter Container startet überhaupt erst, wenn er mindestens eine WP oder eine WB in Influx findet, wie es scheint. Müsste nochmal den Container hochziehen und in die Logs schauen...

gereons, 2024-08-28:
>  Oder sehe ich den Wert einfach nicht, weil ich nicht gut schaue. 😄

Der Wert steht oben als "Autarkie", wenn du nix zu splitten hast.

ledermann, 2024-08-28:
@AlpenFlizzer Korrekt, der derzeitige Power-Splitter erfordert zwingend Wärmepumpe und/oder Wallbox. Wenn es beides nicht gibt, also nur Hausverbrauch vorhanden ist, dann gibt es nichts aufzuteilen. Der Netzbezug ist dann vollständig durch den Hausverbrauch entstanden. Der PV-Anteil entspricht dann der bereits dargestellten Autarkie. Dieses Thema ist aber besser in #3330 aufgehoben.

Die Ausweisung des Batterie-Anteils (also das, was hier in diesem Issue noch offen ist), wäre aber tatsächlich auch ohne Power-Splitter denkbar, also auch, wenn es nur den Hausverbrauch gibt.