solectrus/solectrus #2097: incompatible character encodings

Timokirch, 2023-06-19:
Hi George and community, 

thank you very much for this banger of software. I have installed it yesterday on my proxmox lxc environment and i noticed that I spend more time than I should looking at the dashboard throughout the day. Anyhow I wonder I could get a helping hand here: 

This morning I was greeted by a 500 error and the logs state the following: 
`I, [2023-06-19T05:43:48.073372 #11]  INFO -- : [192.168.2.100] method=GET path=/stats/inverter_power/now format=html controller=StatsController action=index status=500 duration=26.75 view=0.00 user_agent=Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36
F, [2023-06-19T05:43:48.073956 #11] FATAL -- : [192.168.2.100]
[192.168.2.100] ActionView::Template::Error (incompatible character encodings: ASCII-8BIT and UTF-8):
[192.168.2.100]      5: turbo-stream action="morph" target="stats" children-only= true
[192.168.2.100]      6:   template
[192.168.2.100]      7:     - if timeframe.now?
[192.168.2.100]      8:       = render StatsNow::Component.new calculator:, field:
[192.168.2.100]      9:     - else
[192.168.2.100]     10:       = render StatsRange::Component.new calculator:, timeframe:, field:
[192.168.2.100]
[192.168.2.100] app/views/stats/index.turbo_stream.slim:8
[192.168.2.100] app/controllers/stats_controller.rb:6:in `index'
[192.168.2.100] app/controllers/concerns/auto_locale.rb:14:in `switch_locale'
I, [2023-06-19T05:43:48.080267 #11]  INFO -- : [192.168.2.100] method=GET path=/500 format=html controller=ErrorsController action=show status=500 duration=6.04 view=5.86 user_agent=Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/114.0.0.0 Safari/537.36`

Tested on mobile device (iOS Tablet) & laptop (macOS with Safari, Chrome & FF. All devices set to English as OS language.

Thanks for your help in advance and kind regards
Timo

ledermann, 2023-06-19:
Strange, I didn't see that before. There are some places in the source code where special characters like `°C`, `€` or `⇒` are used - but they are already formatted as UTF-8.

Is your installation publicly available so I can do some checks myself?

Otherwise:
- Are you using the official release v0.11.0? 
- Does the error occur on other pages as well? (day/week/month/year/all, top10, settings, login)
- What happens if you change your browser to German (instead of English)?
- Since it worked without problems yesterday: Did you change anything today?

ledermann, 2023-06-19:
Ok, thanks for the details and the URL via e-mail. Just to be sure: Switching one device to German also fixes the problem on the **other** device that is still in English, so now the problem is not reproducible at all. Funny :-)
Maybe caching is involved.

I just made a change that **might** fix the issue. If you want to try it out, you can change the Docker image from `ghcr.io/solectrus/solectrus:latest` to `ghcr.io/solectrus/solectrus:develop`. After restarting the containers, I suggest trying a device configured for English first.

Timokirch, 2023-06-19:
No problem, I am happy that you are supporting :) 

Did the update to v0.11.0-22-gcbda0a47, rebooted, changed languages on several devices after each other or in parallel, tried different browser with different languages on the same machine. Seems to be working.

ledermann, 2023-06-19:
Great, so it seems my fix has solved the issue. Thanks a lot for your feedback!