solectrus/solectrus #2115: Datenbank sichern/kopieren/übertragen

Zeppelin-PV, 2023-06-25:
Ich bin Solectrus User aber kein Profi im Umgang mit Linux, Raspberry und deren OS's
Folgendes:
Ich habe zwei Raspi 4 mit Solectrus am laufen. Einfach wegen der Redundanz. Heute nacht ist einer in der Tat ausgestiegen, warum auch immer. Habe ihn mit einer Kopie des zweiten wieder zum laufen gebracht. Jetzt ist es so, dass während den Kopier und Übertragungszeiten Daten fehlen.
[image]
Der zweite ist ohne diese Unterbrechungen
[image]
Frage:
gibt es eine Möglichkeit diese Lücken zu füllen mit der Datenbank des anderen. Am schönsten wäre es über das Netzwerk.
Oder mittels USB Stick ? Wie ist die richtigen Vorgehensweise ?
Danke jetzt schon für Eure Tipps.

ledermann, 2023-06-25:
Eines vorweg: Ohne Linux-Kenntnisse wirds schwierig.

Grundsätzlich kannst Du eine InfluxDB sichern und wiederherstellen:
https://docs.influxdata.com/influxdb/v2.7/backup-restore/

Nebenbei: Ich habe ein Repo erstellt, das Influx-Backups automatisiert erstellt und nach Amazon S3 schiebt:
https://github.com/solectrus/influxdb2-s3-backup

In Deinem Fall könnte es einfacher sein, den InfluxDB-Ordner zu kopieren. Vorher muss aber unbedingt die betroffene Influx-Datenbank gestoppt werden. Vorsicht, da kann man viel kaputt machen!

Alternativ könntest Du für den betroffenen Tag (bzw. Woche) die CSV-Datei bei mein-senec.de herunterladen und dann den `senec-importer` verwenden:
https://github.com/solectrus/senec-importer

Alles in allem ist es aber viel Aufwand für eine so kleine Datenlücke.

Zeppelin-PV, 2023-06-26:
Du hattest recht, Georg
es lohnt sich nicht wegen insgesamt 25min da etwas über's Knie zu brechen

[image]

Seltsamerweise sind einige Werte sogar höher trotz Unterbrechung; wie kommt das zustande ?
Bedient der SENEC Webserver die beiden Raspi's nicht gleich ?

Vielen Dank für Deine gezeigten Möglichkeiten
... muss mich mit der Linux-Thematik doch noch etwas mehr beschäftigen :(

Zeppelin-PV, 2023-06-26:
der Vollständigkeit halber hier noch mit den SENEC Portal Werten
[image]