solectrus/solectrus #2129: Status wechselt dauernd zwischen Live und Fail

cg089, 2023-06-28:
Hallo,
ich betreibe Solectrus auf einem Proxmox Docker Host.
Ebenfalls auf Proxmox läuft Home Assistant (als mqtt Server), mit evcc.

Der mqtt ist auf Port 1883 ohne SSL eingebunden.

Die Verbindung von solectrus funktioniert an sich, es kommen auch alle Daten rüber.
In der Online Demo unter
https://solectrus.ledermann.dev/house_power/now
zeigt er rechts oben einen Status mit einem grünen Punkt an. Dort steht "entladen".
[image]

In meiner Instanz wechselt dieser Status jedoch permanent im ca. 15-Sekunden-Takt von live auf fail.
[image]
[image]

Daten sind trotzdem alle da.
Der mqtt Host ist vom Docker Host auch permanent erreichbar und anpingbar.

ledermann, 2023-06-29:
Grundsätzlich ist es so, dass die Status-Anzeige `"FAIL"` anzeigt, wenn der letzte Messwert älter ist als 10 Sekunden. Somit ist die Ursache in Deinem Fall vermutlich die, dass das Abrufen der Messwerte mit einem zu großen zeitlichen Intervall stattfindet. Hier müsstest Du beim MQTT-Broker oder bei evcc nachsehen, wie man das Intervall verkürzen kann.

In meiner Installation (https://solectrus.ledermann.dev) wird übrigens anstelle von `"LIVE"` ein konkreter Status als Text angezeigt, also z.B. `"ENTLADEN"`. Ich benutze den SENEC-Collector, der diesen Status vom Speicher abholt.
Der MQTT-Collector kann diesen auch schreiben, hierzu dient die `.env`-Variable `MQTT_TOPIC_CURRENT_STATE`. Dazu muss Deine PV-Anlage diesen Status aber auch liefern können.

cg089, 2023-06-29:
Hi,
super, vielen Dank!
Gibt es eine Doku aller MQTT States? Ich würde gern auch noch die Temperatur hineinpushen.

ledermann, 2023-06-29:
Für die Temperatur: `MQTT_TOPIC_CASE_TEMP`

Liste aller Topics:
https://github.com/solectrus/mqtt-collector/blob/develop/.env.example

StevieC121176, 2023-06-29:
Ich lasse mir jetzt als Status abwechselnd alle 30s die aktuelle Wetterlage und den aktuellen Status meines Wechselrichters anzeigen, mittels iobroker und den mqtt Collector wird der Statustext an Solectrus übergeben. Funktioniert prima.

ledermann, 2023-06-29:
Interessante Lösung :)

Anzeige der Wetterlage (ohne solche Tricks) wäre etwas, das für SOLECTRUS auch interessant wäre. Ich habe mal ein Issue angelegt:
https://github.com/solectrus/solectrus/issues/2132

closed 2023-06-29 by ledermann