solectrus/solectrus #2131: Monats/Quartalsvergleich der einzelnen Jahre

StevieC121176, 2023-06-29:
Wäre es möglich, zusätzlich zum Jahresvergleich auch einen Monats bzw. Quartalsvergleich als Diagramm anzuzeigen? Zur Verdeutlichung habe ich mal 2 Screenshots angehängt. Finde diese Vergleiche der einzelnen Jahre recht spannend.

[image]
[image]
[image]

ledermann, 2026-09-19:
Work in progress ...

[image]

ledermann, 2026-09-29:
Jetzt in der Live-Demo als **Preview** zu sehen:
https://demo.solectrus.de/inverter_power/all/by_month

Fließt in v1.4 ein, hier ein paar Hinweise dazu:
https://github.com/orgs/solectrus/discussions/5966

deThommy, 2026-09-29:
Hallo Georg,

mir gefällt die Ansicht richtig gut. Als Ergänzung: ich fände es hilfreich (oder sagen wir besser interessant), wenn hier noch eine Linie integriert wäre, die den Durchschnittswert visualisiert, so sieht man dann direkt, welche Monate überdurchschnittlich im Jahresvergleich sind. Praktisch als Glockenkurve über das gesamte Diagramm. Anhand des aktuellen Verbrauchs wäre also der jetzige September bei deiner Anlage leicht oberhalb des Durchschnitts

[image]

Ist das verständlich formuliert? 

praktisch wäre das ein Durchschnittswert pro Monats-Kohorte, der dann einfach verbunden wird?!

Liebe Grüße

ledermann, 2026-09-30:
@deThommy Ja, der Durchschnittswert ist sinnvoll, habe das nun ergänzt - aber nicht als Kurve, sondern als gestrichelte Linie je Gruppe. Eine interpolierte Kurve würde nur für die mittlere Säule die Abweichung vom Durchschnitt visualisieren. Daher sieht es nun so aus:
https://demo.solectrus.de/inverter_power/all/by_month

[image]

Zeppelin-PV, 2026-09-30:
hallo Georg - großartig - da kommt ja einiges auf uns zu
war mal bei der Demo neugierig und habe folgendes gesehen
Hier werden Strombezug und Einspeisung getrennt, somit könnte der Text in Klammer entfallen.
Genau das wünschte ich mir in der Strombilanz auch - die separate Darstellung von Bezug und Einspeisung (getrenntes issue ?)

[image]

ledermann, 2026-09-30:
@Zeppelin-PV Um es einheitlich zu haben, habe ich es anders herum gemacht: Bei Netzbezug/Einspeisung sowie Akkubeladung/Entladung erscheinen Diagramme grundsätzlich mit Säulen in beide Richtungen.

[image]

closed 2026-10-12 by ledermann

ledermann, 2026-10-12:
Implementiert in `develop` ab `v1.4-beta-120`.
Weitere Infos in https://github.com/orgs/solectrus/discussions/5966#discussioncomment-18882124