solectrus/solectrus #2132: Wetterlage darstellen

ledermann, 2023-06-29:
Außentemperatur und Wolkenstand sollten über eine Wetter-API ermittelt und dargestellt werden.

StevieC121176, 2023-06-29:
Wolkenstand könnte man evtl. auch durch Icons darstellen, das spart Platz.

StevieC121176, 2023-06-30:
Die Außentemperatur könnte alternativ auch über Mqtt aus dem häuslichen Smarthome übergeben werden ähnlich dem Temperaturwert für den Speicher.

ledermann, 2023-06-30:
Meine Vorstellung hierzu ist, eine API wie z.B. https://open-meteo.com/ abzufragen. Ähnlich dem Forecast-Collector wären dazu

ledermann, 2023-06-30:
Meine Vorstellung hierzu ist, eine API wie z.B. https://open-meteo.com/ abzufragen. Ähnlich dem [Forecast-Collector](https://github.com/solectrus/forecast-collector) könnte das ein eigener Docker-Container erledigen, der natürlich optional wäre und Wetter-Daten nach InfluxDB schreibt. Relevante Daten wären vermutlich Temperatur und WMO code, letzter könnte dann in SOLECTRUS als Icon dargestellt werden.

Aber hey, das Ganze sehe ich als low-prio an :)

StevieC121176, 2023-07-01:
Das so aufzubauen wie den Forecast-Collector finde ich eine gute Idee, dann hat man auch die historischen Daten.

Ich persönlich schaue doch recht oft auf die Wetterdaten, deswegen lasse ich mir die auch in dieser Infozeile anzeigen (jetzt nur die Wetterdaten, keine Wechselrichterinfos mehr (die waren bei meinem Wechselrichter nicht sonderlich relevant)). Allerdings habe ich dabei ein kleines Problem. Die Anzeige wechselt ca. alle 5min von meinem Wetterdaten auf "Live". Nach 5min ca. werden dann wieder für ca. 5 min meine Wetterdaten angezeigt. Der Punkt daneben bleibt allerdings immer grün. Der Datenpunkt im iobroker ist auch immer gefüllt und wird per MQTT bereit gestellt. 
Hat jemand eine Ahnung woran das Wechseln der Anzeige liegen kann?

 
[image]

StevieC121176, 2023-07-02:
Was hier evtl. auch noch interessant wäre ist die Sonnenscheindauer am Tag.
(ich habe dazu die "dasWetter" App dafür genutzt, die gibt solche Infos her.)

  
[image]

Zeppelin-PV, 2023-07-03:
Ich möchte mich in die Wetterdebatte mal einklinken, ohne jemand nahe treten zu wollen.
Solectrus ist ein hervorragendes Tool für die PV-Anlage mit einer Vielzahl von 'Echtzeit'- und Bilanzierungsanzeigen.
Eine Abhängigkeit der PV-Anlage zum Wetter streitet niemand ab, Aber aus Solectrus eine Wetterstation zu machen 
erscheint mir etwas zu weit gegriffen. Es ist ja schon eine Forecast Funktion enthalten die mal mehr oder weniger stimmt.
Das würde auch für alle anderen Wetteranzeigen in Solectrus zutreffen. Wetter ist nunmal Wetter. 
Wenn ich Infos zum Wetter haben möchte gehe ich zu meiner Wetterstation oder schaue in Echtzeit zum Himmel.
Es kommt ja auch vermutlich niemand auf die Idee an seiner Wetterstation die PV-Erzeugung anzeigen zu lassen. 
Sollte das ein- oder andere sinvolle WetterGadget Platz finden in Solectrus, dann ok. Aber es sollte hauptsächlich ein
PV Monitoring bleiben.
Hier ein 'Negativ' Beispiel das nichts bringt ist die Anzeige der SolarmanApp (Balkonkraftwerk) mit den kleinen grauen Wolken
[image]
... konnte ich in Echtzeit mitverfolgen ;-)

ledermann, 2023-07-03:
Ja, @Zeppelin-PV, Dein Einwand ist berechtigt. 

Eine grobe, noch nicht durchdachte Möglichkeit ist die Idee, historische PV-Messwerte mit historischen Wetterdaten zusammenzubringen. So könnte in der Top10-Erzeugung jeweils die Höchsttemperatur der angezeigten 10 Tage mit aufgeführt werden. 

Möglicherweise ist aber auch das sinnfrei. Auf jeden Fall ist das kein Thema, das ich bevorzugt in absehbarer Zeit umsetzen möchte. Da gibt es zweifellos wichtigere Dinge.

Denoch finde ich die von @StevieC121176 erarbeitete Integration von Wetter-Informationen in den Status von SOLECTRUS eine witzige Sache :)

ledermann, 2024-02-21:
Nur mal so als Erinnerung für mich selbst: Nachdem hier eine Wärmepumpe im Einsatz ist, sehe ich weiteren Bedarf für Wetterinformationen.

Der Stromverbrauch der Wärmepumpe hängt ja sehr von der Außentemperatur ab. Es wäre vielleicht nützlich, wenn SOLECTRUS darstellen würde, wie die durchschnittliche Außentemperatur eines gewählten Zeitraums mit dem Stromverbrauch der Wärmepumpe zusammenhängt. Das könnte z.B. dadurch erreicht werden, dass im Diagramm des Wärmepumpen-Verbrauchs zusätzlich eine Temperaturkurve dargestellt wird.

StevieC121176, 2024-02-21:
Das ist eine spannende Sache. Vielleicht könnte man auch hier die Daten per mqtt Collector bereitstellen. Ich habe z.bsp. Noch 2 Klimaanlagen, wo ich den Stromverbrauch per mqtt abgreifen kann und dann auch in Verbindung mit der Temperatur darstellen könnte.

ledermann, 2024-02-21:
Wetterdaten per MQTT oder Klimaanlagen-Messwerte per MQTT? Ich dachte hier eigentlich daran, die Wetterdaten über einen separaten Collector von OpenMeteo o.ä. abzufragen, also ohne MQTT.

Der MQTT-Collector könnte beides aber auch erledigen, den müsste ich sowieso mal aufpolieren.

StevieC121176, 2024-02-21:
Ich dachte eigentlich beides über mqtt. Wetterdaten direkt von einem Wetterdienstanbieter wäre natürlich noch besser.

patricknitsch, 2025-11-26:
Ist ne super Sache mit der Außentemperatur. Leider nur sichtbar, wenn man eine Wärmepumpe konfiguriert hat 🙂

ledermann, 2025-11-29:
@hj-hefer Die Seite [/forecast](https://demo.solectrus.de/forecast) hat mit der Wärmepumpe nichts zu tun. Sie benötigt nur die Sensoren `INVERTER_POWER_FORECAST` und optional `OUTDOOR_TEMP_FORECAST`. Wo diese herkommen, ist egal.