solectrus/solectrus #2138: Kosten / Einsparung - schwäbische Anzeige

Zeppelin-PV, 2023-07-01:
hallo Georg
könnte man den Schwaben zuliebe bei der Kosten/Einsparung Anzeige auch die zwei Kommastellen Cent anzeigen ?
... auch den NichtSchwaben zuliebe
LG
Ernst /Zeppelin-PV

ledermann, 2023-07-01:
Mir ist nicht ganz klar, was Du hier meinst. Grundsätzlich ist es so, dass SOLECTRUS bei großen Zahlen (kWh oder €) die Nachkommastellen weglässt. Das halte ich eigentlich auch für richtig und sinnvoll.

Hier ein Beispiel, [Monat Juni 2023 aus der Live-Demo](https://solectrus.ledermann.dev/inverter_power/2023-06):

[image]

Die Kosten sind hier sehr gering, daher werden sie **mit** Nachkommastellen angegeben. Die Einsparung ist hingegen sehr hoch, daher werden die Cents **nicht** angegeben. 

Ich möchte Zahlenwüsten vermeiden. Ansonsten denke ich bei dem Thema immer an meinen alten Physik-Lehrer, der meinte: "Suggeriere keine Genauigkeit durch viele Nachkommastellen, wenn Du gar nicht so genau messen kannst!"

Zeppelin-PV, 2023-07-01:
bis Monat sind die Cents da, im Jahr und Gesamt nicht, aber ...
... Dein Physik-Lehrer hat recht !!!
Issue schließen

ledermann, 2023-07-01:
Ok, wir belassen es so.

Nur der Vollständigkeit halber: Es hat nichts mit dem Zeitraum zu tun, sondern nur mit der Zahl. Unter 10 € gibt es Nachkommastellen, ansonsten nicht.

closed 2023-07-01 by ledermann