solectrus/solectrus #2140: Keine Daten

spyro2000, 2023-07-01:
Hallo zusammen,

habe heute Solectrus auf meinem Pi (3B) aufgesetzt und soweit hat auch alles gut funktoniert. Es laufen offenbar auch munter Daten ein:

```
solectrus-senec-collector-1     | Got record #66 from SENEC at 192.168.178.158: AKKU VOLL, Inverter 1488 W, House 249 W, 2023-07-01 14:28:59 +0000
solectrus-senec-collector-1     | Successfully pushed record #64 to InfluxDB
solectrus-senec-collector-1     | Successfully pushed record #63 to InfluxDB
solectrus-senec-collector-1     | Successfully pushed record #65 to InfluxDB
solectrus-senec-collector-1     | Successfully pushed record #66 to InfluxDB
````

Allerdings heißt es im Webinterface trotzdem immer KEINE DATEN und der Status ist 'FAIL'.

Mein .env file:

```
##################################################################
# Dashboard application (the main part)
#
# Domain name or IP address of your host
APP_HOST=192.168.178.87
#
# SSL redirect: Use "true" (which is the default) if you want to auto-redirect to https
FORCE_SSL=false
#
# Secret token to secure cookies, 128 chars long hexadecimal encoded string (don't use this example, make some random changes!)
# Currently there are no cookies in Soletrus, but this may change in the future
SECRET_KEY_BASE=f60debe97dcb73280a2cc83668fd60e8d0e8e48269036a7bce980ee53cfb312e377989a750b8c945a5f69b041289ecb4e2d9e40641b81257c65ac2d43e3c837f
#
# Date of commissioning of your photovoltaic system
INSTALLATION_DATE=2023-06-28
#
# Price you pay for 1kWH (in EUR)
# Can be changed later in the UI when logged in as administrator
ELECTRICITY_PRICE=0.32
#
# Price you get for 1kWH (in EUR)
# Can be changed later in the UI when logged in as administrator
FEED_IN_TARIFF=0.0848
#
# Password for the PostgreSQL database, used by the app to comunicate internally with the database
POSTGRES_PASSWORD=my-secret-db-password
#
# Password to login as administrator, required to manage settings like historical prices
ADMIN_PASSWORD=my-secret-login-password

##################################################################
# Influx database settings
#
# Influx host (to access from SOLECTRUS dashboard and collectors)
INFLUX_HOST=influxdb
INFLUX_SCHEMA=http
INFLUX_PORT=8086
#
# Credentials for the Influx database, don't change after the first run!
INFLUX_ORG=solectrus
INFLUX_USERNAME=admin
INFLUX_PASSWORD=ExAmPl3PA55W0rD
INFLUX_ADMIN_TOKEN=my-super-secret-admin-token
#
# Set these names before first run, they can't be changed later!
INFLUX_BUCKET=my-solectrus-bucket
INFLUX_MEASUREMENT_PV=my-pv-measurement
INFLUX_MEASUREMENT_FORECAST=my-forecast-measurement
#
# To keep things simple, we use ONE token (INFLUX_ADMIN_TOKEN) for both writing and reading.
# For better security, you can use two separate tokens, created via the InfluxDB frontend.
INFLUX_TOKEN_WRITE=my-super-secret-admin-token
INFLUX_TOKEN_READ=my-super-secret-admin-token
#
# Volume path for storing the Influx data
INFLUX_VOLUME_PATH=/home/spyro/solectrus/influxdb

##################################################################
# REDIS settings
#
REDIS_VOLUME_PATH=/home/spyro/solectrus/redis

##################################################################
# PostgreSQL database settings
#
DB_VOLUME_PATH=/home/spyro/solectrus/postgresql

##################################################################
# SENEC Collector
#
# Change this to your own SENEC IP address!
SENEC_HOST=192.168.178.158
SENEC_INTERVAL=5

##################################################################
# Solar forecasting with https://forecast.solar
# API docs: https://doc.forecast.solar/doku.php?id=api:estimate
#
# Latitude of the plant location
FORECAST_LATITUDE=48.456167
#
# Longitude of the plant location
FORECAST_LONGITUDE=11.140888
#
# Plane declination: 0 (horizontal) - 90 (vertical)
FORECAST_DECLINATION=20
#
# Plane azimuth: -180 ... 180 (-180 = north, -90 = east, 0 = south, 90 = west, 180 = north)
FORECAST_AZIMUTH=0
#
# Installed modules power in kilowatt peak (kWp)
FORECAST_KWP=7.9
#
# Update interval in seconds, 900s = 15 minutes, the public (and free) API allows a minimum of 900 seconds
FORECAST_INTERVAL=900
````
Liegt es daran, dass mein user nicht 'pi' heißt sondern 'spyro'? Ist das irgendwo noch hardcoded...?

ledermann, 2023-07-01:
Danke für die Beschreibung, hier offenbart sich ein Fehler meinerseits. Das Customizing des InfluxDB-Measurement wurde erst kürzlich eingeführt, der SENEC-Collector berücksichtigt das aber noch gar nicht. Das muss ich noch ändern.

Kurzfristige Abhilfe: Ändere in der `.env` die Variable `INFLUX_MEASUREMENT_PV` von `my-pv-measurement` auf `SENEC`. Dann sollte es klappen.

Ansonsten noch:
- Der Pi-Name spielt keine Rolle.
- Das Passwort für den Admin-Login ist das, das in der `.env` als `ADMIN_PASSWORD` steht.

spyro2000, 2023-07-01:
Danke dir! Ein Neustart später (5 Min) hat's dann funktioniert. Ich liebe es! :)