solectrus/solectrus #2150: Support für Dynamische Tarife

teremok87, 2023-07-06:
Es wäre toll, wenn man dynamische Tarife wie Tibber oder Awattar für die Berechnung der Kosten hinterlegen könnte. 

Vielen Dank im Voraus!!
Gruß aus Bayern

ledermann, 2023-07-06:
Grundsätzlich ist das ein hochinteressantes Thema, das ich gerne angehen würde. Momentan halten mich aber zwei Dinge davon ab:

- In der aktuellen Version ist SOLECTRUS dafür ausgelegt, dass sich ein Strompreis eher selten ändert. Die Preise stehen in der PostgreSQL-Datenbank und es findet ein ausgetüfteltes Verfahren statt, um präzise die Kosten und Erträge für eine zeitliche Periode zu ermitteln. Bei sich stündlich ändernden Preisen wäre das jedoch sehr ineffizient und würde zu Performance-Problemen in der Darstellung führen. Die Preise müssten stattdessen in InfluxDB stehen, genau wie die Messwerte. Dann müssten aber alle Berechnungen ganz anders laufen, das wäre ein größerer Umbau.

- Ich persönlich bin derzeitig noch an einen Old-School-Tarif gebunden. Sobald der ausläuft, könnte ich mir einen Wechsel zu einem Anbieter von dynamischen Tarifen vorstellen. Anschließend würde auch die Motivation steigen, meine Freizeit in die Entwicklung eines solchen Features zu investieren ;-)

teremok87, 2024-01-08:
Hallo Georg,

wie sieht es jetzt mit deiner Motivation aus? Hast du noch Interrese an einem Dynamischen Stromtarif? Ich Steige im Februar auf Tibber um. Ich würde gerne weiterhin Solectrus nutzen. ;)

Übrigens, hast du darüber nachgedacht ein ähnliches Lizenzmodel zu machen, wie es die Kollegen bei EVCC machen.
Ich Zahle da gerne 2$ im Monant als Github abo. Das würde ich auch in deinem Fall machen. Denke darüber nach :)

Gruß
Artem

ledermann, 2024-01-08:
@teremok87 Definitiv habe ich daran Interesse, hinter den Kulissen wird daran auch schon gearbeitet. Das Abrufen der Preise von Tibber und schreiben in die Datenbank funktioniert auch bereits. Das ist aber nur die halbe Miete.

Der aufwendigste Teil ist nämlich der erforderliche komplette Umbau der Kosten- und Erdparnisberechnung. Auch hier habe ich bereits einiges implementiert, aber das ist noch nicht fertig. Die Stromkosten müssen stundenweise ermittelt und dann aggregiert nach Tagen abgelegt werden. Wenn das fertig ist, dann ist SOLECTRUS wirklich Tibber-kompatibel.

Zum Lizenzmodel: Ja, auch dazu mache ich mir viele Gedanken. Eine Premium-Version von SOLECTRUS, die per Abo bezahlt wird, liegt im Bereich des Möglichen. Hierzu werde ich genaueres mitteilen, wenn es wirklich spruchreif ist.

WolfImBusch, 2024-09-08:
Einen schönen guten Morgen @ledermann ,

auch ich werde demnächst zu Tibber wechseln und bin dementsprechend sehr an einem Support für dynamische Tarife  interessiert. Ich kann natürlich nachvollziehen, dass die Implementierung sicher nicht trivial sein wird. Trotzdem möchte ich es hier erwähnen, da Du vielleicht auch ein Interesse daran haben könntest zu erfahren, welches Features sich Deine "Community" wünscht ... :-)

Damit die Berechnung der Kosten in Solectrus auch zukünftig einigermaßen einigermaßen aussagekräftig sind, würde ich den Durchschnitsspreis/Monat von Tibber in Solecturs eintragen, aber ich natürlich nicht ganz soooo schön ... oder hast Du eine bessere Idee?!?

Ich danke Dir und viele Grüße

ledermann, 2024-09-09:
@WolfImBusch Danke für dein Feedback! Die Implementierung ist ein komplexes Thema, da ein ganz anderer Ansatz verfolgt werden muss. Ich habe aber bereits ein grobes Konzept, wie das machbar ist. Wann ich das implementieren werde, stehe aber noch nicht fest. Bis dahin ist das Erfassen von einem Durchschnittspreis für jeden Monat ein denkbarer Kompromiss.

WolfImBusch, 2024-11-30:
Hi @ledermann ,

total nachvollziehbar - sogar für mich als Hobby-Coder ... 😄 Zudem zeigt ja auch Deine Umfrage, dass sich Deine Community momentan andere Dinge wünscht, an denen Du gerade dran bist ... 

Was wäre, wenn Du im Rahmen des nächsten Updates die dynamischen Strompreise von (in meinem Fall Tibber 😄) schonmal in der InfluxDB wegspeicherst und in einem kommenden Update die dort abgespeicherten Preise dann wieder verwendest, um bspw. die ganz wirklich ehrliche Ersparnis zu berechnen? 😄

Viele Grüße aus Stuttgart

WolfImBusch, 2024-11-30:
Super!!!! Du bist zu schnell für mich!!! 🤪 Warum auch immer dachte ich, dass das nur in Verbindung mit dem SENEC System möglich ist - aber ich war wohl falsch unterwegs!! Herzlichen Dank und viele Grüße!!!

WolfImBusch, 2024-11-30:
Hallo @ledermann ,

hab mich zu früh gefreut .. :-( Ich habe den Tibber-Collector installiert und konfiguriert. Leider will er nicht :-( (s. Anhang ein Screenshot der Logs auf meiner Synology - entschuldige bitte die unprofessionelle Beistelliung des Logs :-( Man sieht, dass der Container gestartet wird er die Preise erfolgreich runterlädt - danach scheint es ein Problem zu geben - vermutlich beim Parsen der Daten ...
Darf ich Dich um Deine scharfsinnige Analyse und Hilfe bitten? Herzlichen Dank!

[image]
[image]
[image]

Herzlichen Dank und viele Grüße!

hj-hefer, 2024-12-15:
Werde demnächst nach einem Anbieter wechseln der 2 unterschiedliche kWh Preise pro Tag hat (zu festen Zeiten)

Wäre das möglich das ich diese beiden Preise (Nachts von 0-5 Uhr Preis x und die restliche Zeit Preis Y eintragen kann?

Mit freundlichen Grüßen 
Hefer

ledermann, 2024-12-15:
@hj-hefer Das wäre wohl ein **variabler** Tarif, im Gegensatz zum **dynamischen** Tarif wie bei Tibber.

Derzeitig geht beides in SOLECTRUS leider nicht. Es sind nur Fixpreis-Tarife möglich, also **ein** kWh-Preis, der über einen längeren Zeitraum konstant bleibt. Langfristig möchte ich das aber ändern.

stefanriegel, 2025-02-18:
würde gerne auf Solectus umsteigen, gibts schon news zu Tibber Integration?

ledermann, 2025-02-18:
@stefanriegel Ich möchte das zukünftig gerne realisieren. Wann das sein wird, weiß ich aber selber noch nicht. Weitere News gibt es leider nicht dazu.

Wenn du bei der Installation von SOLECTRUS den [Tibber-Collector](https://docs.solectrus.de/referenz/tibber-collector/) von Anfang an installierst und somit deine Strompreise mitschreibst, dann werden diese Daten bei Verfügbarkeit dieses Features zur Berechnung herangezogen, also auch rückwirkend.